THLfi/read.gt3x #3: Reading old GT3X format

muschellij2, 2019-08-20:
For previous formats (https://github.com/actigraph/NHANES-GT3X-File-Format), for example from https://help.theactigraph.com/entries/21688392-GT3X-ActiSleep-Sample-Data, or the attached
[GT3X+ (01 day).gt3x.zip](https://github.com/THLfi/read.gt3x/files/3522749/GT3X%2B.01.day.gt3x.zip)

# Using file from GitHub

This is an older format from https://github.com/actigraph/NHANES-GT3X-File-Format
but it was uploaded for this issue

## Download data

``` r
library(read.gt3x)
url = "https://github.com/THLfi/read.gt3x/files/3522749/GT3X%2B.01.day.gt3x.zip"
destfile = tempfile(fileext = ".zip")
dl = download.file(url, destfile = destfile)
```

## Unzip file

``` r
gt3x_file = unzip(destfile, exdir = tempdir())
gt3x_file = gt3x_file[!grepl("__MACOSX", gt3x_file)]
gt3x_file
#> [1] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/GT3X+ (01 day).gt3x"
is_gt3x(gt3x_file)
#> /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/GT3X+ (01 day).gt3x 
#>                                                                             TRUE
```

Here we can try to read the file and see the error

``` r
res = read.gt3x(gt3x_file, verbose = TRUE)
#> Input is a .gt3x file, unzipping to a temporary location first...
#> Unzipping gt3x data to /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok
#> 1/1
#> Unzipping /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/GT3X+ (01 day).gt3x
#> /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/GT3X+ (01 day).gt3x doesn't contain log.bin
#> /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/GT3X+ (01 day).gt3x did not contain both log.bin and info.txt. Unzipping failed.
#> Error in result_paths[i] <- unzip_single_gt3x(gt3xfiles[i], location = location, : replacement has length zero
```

Let’s look at the files inside this gt3x (old format):

``` r
unzipped = unzip(gt3x_file, exdir = tempdir())
unzipped
#>  [1] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/activity.bin"
#>  [2] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/battery.bin" 
#>  [3] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/eeprom.bin"  
#>  [4] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/firmware.bin"
#>  [5] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/gt3xplus.ico"
#>  [6] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/info.txt"    
#>  [7] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/log.txt"     
#>  [8] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/lux.bin"     
#>  [9] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/metadata"    
#> [10] "/var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/ram.bin"
```

# What if we tried activity.bin

If we renamed `activity.bin` to `log.bin`, does this work?

``` r
activity = unzipped[ basename(unzipped) == "activity.bin"]
# rename to log.bin
log_file =  file.path(dirname(activity), "log.bin")
file.copy(activity,log_file, overwrite = TRUE)
#> [1] TRUE
```

Must rezip the file with files at the root

``` r
owd = getwd()
# zipping needs paths to start at the root of the zip
setwd(tempdir())
new_gt3x = tempfile(fileext = ".gt3x")
zip(new_gt3x, files = basename(c(unzipped, log_file)))
setwd(owd)
```

Seems like this doesn’t work as the format is different

``` r
res = read.gt3x(new_gt3x, verbose = TRUE)
#> Input is a .gt3x file, unzipping to a temporary location first...
#> Unzipping gt3x data to /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok
#> 1/1
#> Unzipping /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/file55246f9fbdb2.gt3x
#>  === info.txt and log.bin extracted to /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpBJtmok/file55246f9fbdb2
#> GT3X information
#>  $ Serial Number     :"NEO1DXXXXXXXX"
#>  $ Device Type       :"GT3XPlus"
#>  $ Firmware          :"2.5.0"
#>  $ Battery Voltage   :"4.22"
#>  $ Sample Rate       :30
#>  $ Start Date        : POSIXct, format: "2012-06-27 10:54:00"
#>  $ Stop Date         : POSIXct, format: "2012-06-28 11:54:00"
#>  $ Download Date     : POSIXct, format: "2012-06-28 16:25:52"
#>  $ Board Revision    :"4"
#>  $ Unexpected Resets :"0"
#>  $ Sex               :"Male"
#>  $ Height            :"172.72"
#>  $ Mass              :"69.8532249799612"
#>  $ Age               :"43"
#>  $ Race              :"White / Caucasian"
#>  $ Limb              :"Ankle"
#>  $ Side              :"Left"
#>  $ Dominance         :"Non-Dominant"
#>  $ DateOfBirth       :"621132192000000000"
#>  $ Subject Name      :"GT3XPlus"
#>  $ Last Sample Time  : 'POSIXct' num(0) 
#>  - attr(*, "tzone")= chr "GMT"
#>  $ Acceleration Scale:
#> Parsing GT3X data via CPP.. expected sample size:
#> Error in parseGT3X(logpath, max_samples = samples, scale_factor = info$`Acceleration Scale`, : Expecting a single value: [extent=0].
```

<sup>Created on 2019-08-20 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0)</sup>

<details>

<summary>Session info</summary>

``` r
devtools::session_info()
#> ─ Session info ──────────────────────────────────────────────────────────
#>  setting  value                       
#>  version  R version 3.6.0 (2019-04-26)
#>  os       macOS Mojave 10.14.6        
#>  system   x86_64, darwin15.6.0        
#>  ui       X11                         
#>  language (EN)                        
#>  collate  en_US.UTF-8                 
#>  ctype    en_US.UTF-8                 
#>  tz       America/New_York            
#>  date     2019-08-20                  
#> 
#> ─ Packages ──────────────────────────────────────────────────────────────
#>  package     * version    date       lib source                           
#>  assertthat    0.2.1      2019-03-21 [1] CRAN (R 3.6.0)                   
#>  backports     1.1.4      2019-04-10 [1] CRAN (R 3.6.0)                   
#>  callr         3.3.1      2019-07-18 [1] CRAN (R 3.6.0)                   
#>  cli           1.1.0      2019-03-19 [1] CRAN (R 3.6.0)                   
#>  crayon        1.3.4      2017-09-16 [1] CRAN (R 3.6.0)                   
#>  desc          1.2.0      2019-07-10 [1] Github (muschellij2/desc@b0c374f)
#>  devtools      2.1.0      2019-07-06 [1] CRAN (R 3.6.0)                   
#>  digest        0.6.20     2019-07-04 [1] CRAN (R 3.6.0)                   
#>  evaluate      0.14       2019-05-28 [1] CRAN (R 3.6.0)                   
#>  fs            1.3.1      2019-05-06 [1] CRAN (R 3.6.0)                   
#>  glue          1.3.1      2019-03-12 [1] CRAN (R 3.6.0)                   
#>  highr         0.8        2019-03-20 [1] CRAN (R 3.6.0)                   
#>  htmltools     0.3.6      2017-04-28 [1] CRAN (R 3.6.0)                   
#>  knitr         1.24       2019-08-08 [1] CRAN (R 3.6.0)                   
#>  magrittr      1.5        2014-11-22 [1] CRAN (R 3.6.0)                   
#>  memoise       1.1.0      2017-04-21 [1] CRAN (R 3.6.0)                   
#>  pkgbuild      1.0.3      2019-03-20 [1] CRAN (R 3.6.0)                   
#>  pkgload       1.0.2      2018-10-29 [1] CRAN (R 3.6.0)                   
#>  prettyunits   1.0.2      2015-07-13 [1] CRAN (R 3.6.0)                   
#>  processx      3.4.1      2019-07-18 [1] CRAN (R 3.6.0)                   
#>  ps            1.3.0      2018-12-21 [1] CRAN (R 3.6.0)                   
#>  R6            2.4.0      2019-02-14 [1] CRAN (R 3.6.0)                   
#>  Rcpp          1.0.2      2019-07-25 [1] CRAN (R 3.6.0)                   
#>  read.gt3x   * 0.1.0      2019-08-15 [1] local                            
#>  remotes       2.1.0      2019-06-24 [1] CRAN (R 3.6.0)                   
#>  rlang         0.4.0      2019-06-25 [1] CRAN (R 3.6.0)                   
#>  rmarkdown     1.14       2019-07-12 [1] CRAN (R 3.6.0)                   
#>  rprojroot     1.3-2      2018-01-03 [1] CRAN (R 3.6.0)                   
#>  sessioninfo   1.1.1      2018-11-05 [1] CRAN (R 3.6.0)                   
#>  stringi       1.4.3      2019-03-12 [1] CRAN (R 3.6.0)                   
#>  stringr       1.4.0      2019-02-10 [1] CRAN (R 3.6.0)                   
#>  testthat      2.1.1      2019-04-23 [1] CRAN (R 3.6.0)                   
#>  usethis       1.5.1.9000 2019-08-15 [1] local                            
#>  withr         2.1.2      2018-03-15 [1] CRAN (R 3.6.0)                   
#>  xfun          0.8        2019-06-25 [1] CRAN (R 3.6.0)                   
#>  yaml          2.2.0      2018-07-25 [1] CRAN (R 3.6.0)                   
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

muschellij2, 2019-08-21:
It seems as though if you use `activity.bin`, you should be able to use `parseActivity` from https://github.com/THLfi/read.gt3x/blob/master/src/parseGT3X.cpp#L186 and use this on the file directly.  Would that be possible to open that function to the API?

TuomoNieminen, 2019-08-23:
This is fixed in #2 , correct?

muschellij2, 2019-08-23:
Correct.
Best,
John

TuomoNieminen, 2019-08-23:
Great, thanks. Closing this issue.

closed 2019-08-23 by TuomoNieminen