THLfi/read.gt3x #4: Time shifted 1s

muschellij2, 2019-09-19:
Time is shifted by 1s as the first `time_index` is not `0` when parsing the `read.gt3x` as noted by @martakarass

Expected behavior: the first time stamp is the start time from the header.

## Download the Data

``` r
library(read.gt3x)
options(digits = 10, digits.secs = 5)
gt3xfile <- gt3x_datapath(1)
#> Downloading gt3x sample data from https://github.com/THLfi/read.gt3x/releases/download/v1.0/EE_left_29.5.2017-05-30.gt3x.zip
```

## Read in the Data

``` r
X <- read.gt3x(gt3xfile)
```

Getting the attributes of the start time and `time_index`, which is used to create time stamps

``` r
ti = attributes(X)$time_index
ti[1]
#> [1] 100
st = attributes(X)$start_time
st
#> [1] "2017-05-23 09:15:00 GMT"

## Create `data.frame`
df = as.data.frame(X)
df$time[1]
#> [1] "2017-05-23 09:15:01 GMT"
df$time[1] - st
#> Time difference of 1 secs
```

<sup>Created on 2019-09-19 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0)</sup>

<details>

<summary>Session info</summary>

``` r
devtools::session_info()
#> ─ Session info ──────────────────────────────────────────────────────────
#>  setting  value                       
#>  version  R version 3.6.0 (2019-04-26)
#>  os       macOS Mojave 10.14.6        
#>  system   x86_64, darwin15.6.0        
#>  ui       X11                         
#>  language (EN)                        
#>  collate  en_US.UTF-8                 
#>  ctype    en_US.UTF-8                 
#>  tz       America/New_York            
#>  date     2019-09-19                  
#> 
#> ─ Packages ──────────────────────────────────────────────────────────────
#>  package     * version    date       lib
#>  assertthat    0.2.1      2019-03-21 [1]
#>  backports     1.1.4      2019-04-10 [1]
#>  callr         3.3.1      2019-07-18 [1]
#>  cli           1.1.0      2019-03-19 [1]
#>  crayon        1.3.4      2017-09-16 [1]
#>  desc          1.2.0      2019-07-10 [1]
#>  devtools      2.2.0.9000 2019-09-10 [1]
#>  digest        0.6.20     2019-07-04 [1]
#>  DT            0.8        2019-08-07 [1]
#>  ellipsis      0.2.0.1    2019-07-02 [1]
#>  evaluate      0.14       2019-05-28 [1]
#>  fs            1.3.1      2019-05-06 [1]
#>  glue          1.3.1      2019-03-12 [1]
#>  highr         0.8        2019-03-20 [1]
#>  htmltools     0.3.6      2017-04-28 [1]
#>  htmlwidgets   1.3        2018-09-30 [1]
#>  knitr         1.24.3     2019-08-28 [1]
#>  magrittr      1.5        2014-11-22 [1]
#>  memoise       1.1.0      2017-04-21 [1]
#>  pkgbuild      1.0.5      2019-08-26 [1]
#>  pkgload       1.0.2      2018-10-29 [1]
#>  prettyunits   1.0.2      2015-07-13 [1]
#>  processx      3.4.1      2019-07-18 [1]
#>  ps            1.3.0      2018-12-21 [1]
#>  R6            2.4.0      2019-02-14 [1]
#>  Rcpp          1.0.2      2019-07-25 [1]
#>  read.gt3x   * 0.1.0      2019-09-19 [1]
#>  remotes       2.1.0      2019-06-24 [1]
#>  rlang         0.4.0      2019-06-25 [1]
#>  rmarkdown     1.15       2019-08-21 [1]
#>  rprojroot     1.3-2      2018-01-03 [1]
#>  sessioninfo   1.1.1      2018-11-05 [1]
#>  stringi       1.4.3      2019-03-12 [1]
#>  stringr       1.4.0      2019-02-10 [1]
#>  testthat      2.2.1      2019-07-25 [1]
#>  usethis       1.5.1.9000 2019-08-15 [1]
#>  withr         2.1.2      2018-03-15 [1]
#>  xfun          0.9        2019-08-21 [1]
#>  yaml          2.2.0      2018-07-25 [1]
#>  source                            
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  Github (muschellij2/desc@b0c374f) 
#>  Github (r-lib/devtools@d7f0915)   
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  Github (muschellij2/knitr@abcea3d)
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  local                             
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  local                             
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

muschellij2, 2019-09-19:
Expanding this a bit, we show imputing zeroes seems to have off time stamps as well:

## Download the Data

``` r
library(read.gt3x)
options(digits = 10, digits.secs = 5)
gt3xfile <- gt3x_datapath(1)
#> Downloading gt3x sample data from https://github.com/THLfi/read.gt3x/releases/download/v1.0/EE_left_29.5.2017-05-30.gt3x.zip
```

## Read in the Data

``` r
X <- read.gt3x(gt3xfile)
```

Getting the attributes of the start time and `time_index`, which is used to create time stamps

``` r
ti = attributes(X)$time_index
ti[1]
#> [1] 100
st = attributes(X)$start_time
st
#> [1] "2017-05-23 09:15:00 GMT"

## Create `data.frame`
df = as.data.frame(X)
range(df$time)
#> [1] "2017-05-23 09:15:01.00 GMT" "2017-05-30 09:49:03.99 GMT"
```

# Imputing Zeros

Here we impute zeroes, which causes data well outside the range of time
that we should see

``` r
X <- read.gt3x(gt3xfile, imputeZeroes = TRUE)
```

Getting the attributes of the start time and `time_index`, which is used to create time stamps

``` r
ti = attributes(X)$time_index
ti[1]
#> [1] 100
st = attributes(X)$start_time
st
#> [1] "2017-05-23 09:15:00 GMT"

## Create `data.frame`
zdf = as.data.frame(X)
```

The max has changed significantly (over 3 months).

``` r
max(zdf$time)
#> [1] "2017-09-10 14:40:54.58 GMT"
range(zdf$time)
#> [1] "2016-11-01 18:05:07.68 GMT" "2017-09-10 14:40:54.58 GMT"
```

Note how the range of the data now seems to span over an entire year.
<sup>Created on 2019-09-19 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0)</sup>

<details>

<summary>Session info</summary>

``` r
devtools::session_info()
#> ─ Session info ──────────────────────────────────────────────────────────
#>  setting  value                       
#>  version  R version 3.6.0 (2019-04-26)
#>  os       macOS Mojave 10.14.6        
#>  system   x86_64, darwin15.6.0        
#>  ui       X11                         
#>  language (EN)                        
#>  collate  en_US.UTF-8                 
#>  ctype    en_US.UTF-8                 
#>  tz       America/New_York            
#>  date     2019-09-19                  
#> 
#> ─ Packages ──────────────────────────────────────────────────────────────
#>  package     * version    date       lib
#>  assertthat    0.2.1      2019-03-21 [1]
#>  backports     1.1.4      2019-04-10 [1]
#>  callr         3.3.1      2019-07-18 [1]
#>  cli           1.1.0      2019-03-19 [1]
#>  crayon        1.3.4      2017-09-16 [1]
#>  desc          1.2.0      2019-07-10 [1]
#>  devtools      2.2.0.9000 2019-09-10 [1]
#>  digest        0.6.20     2019-07-04 [1]
#>  DT            0.8        2019-08-07 [1]
#>  ellipsis      0.2.0.1    2019-07-02 [1]
#>  evaluate      0.14       2019-05-28 [1]
#>  fs            1.3.1      2019-05-06 [1]
#>  glue          1.3.1      2019-03-12 [1]
#>  highr         0.8        2019-03-20 [1]
#>  htmltools     0.3.6      2017-04-28 [1]
#>  htmlwidgets   1.3        2018-09-30 [1]
#>  knitr         1.24.3     2019-08-28 [1]
#>  magrittr      1.5        2014-11-22 [1]
#>  memoise       1.1.0      2017-04-21 [1]
#>  pkgbuild      1.0.5      2019-08-26 [1]
#>  pkgload       1.0.2      2018-10-29 [1]
#>  prettyunits   1.0.2      2015-07-13 [1]
#>  processx      3.4.1      2019-07-18 [1]
#>  ps            1.3.0      2018-12-21 [1]
#>  R6            2.4.0      2019-02-14 [1]
#>  Rcpp          1.0.2      2019-07-25 [1]
#>  read.gt3x   * 0.1.0      2019-09-19 [1]
#>  remotes       2.1.0      2019-06-24 [1]
#>  rlang         0.4.0      2019-06-25 [1]
#>  rmarkdown     1.15       2019-08-21 [1]
#>  rprojroot     1.3-2      2018-01-03 [1]
#>  sessioninfo   1.1.1      2018-11-05 [1]
#>  stringi       1.4.3      2019-03-12 [1]
#>  stringr       1.4.0      2019-02-10 [1]
#>  testthat      2.2.1      2019-07-25 [1]
#>  usethis       1.5.1.9000 2019-08-15 [1]
#>  withr         2.1.2      2018-03-15 [1]
#>  xfun          0.9        2019-08-21 [1]
#>  yaml          2.2.0      2018-07-25 [1]
#>  source                            
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  Github (muschellij2/desc@b0c374f) 
#>  Github (r-lib/devtools@d7f0915)   
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  Github (muschellij2/knitr@abcea3d)
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  local                             
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  local                             
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

TuomoNieminen, 2019-09-20:
Thanks,

The time shifting by 1s should be easy to fix in the the data frame conversion.

There seems to be something weird going on with the zero imputation feature.

muschellij2, 2019-09-20:
I looked has 30hz data and I think it has the same issue. I can send update
later today

muschellij2, 2019-09-20:
[GT3X+ (01 day).zip](https://github.com/THLfi/read.gt3x/files/3637478/GT3X%2B.01.day.zip)

Here's an example with the 30Hz:

``` r
library(read.gt3x)
url = "https://github.com/THLfi/read.gt3x/files/3637478/GT3X%2B.01.day.zip"
destfile = tempfile(fileext = ".gt3x")
download.file(url, destfile = destfile)

res = read.gt3x(destfile)
at = attributes(res)
at$time_index[1:10]
#>  [1]  0  3  7 10 13 17 20 23 27 30
info = parse_gt3x_info(destfile)
df = as.data.frame(res)
as.numeric(df$time[1:10])%%1
#>  [1] 0.0000000 0.0999999 0.2333333 0.3333333 0.4333334 0.5666666 0.6666667
#>  [8] 0.7666667 0.9000001 0.0000000
```

<sup>Created on 2019-09-20 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0)</sup>

<details>

<summary>Session info</summary>

``` r
devtools::session_info()
#> ─ Session info ──────────────────────────────────────────────────────────
#>  setting  value                       
#>  version  R version 3.6.0 (2019-04-26)
#>  os       macOS Mojave 10.14.6        
#>  system   x86_64, darwin15.6.0        
#>  ui       X11                         
#>  language (EN)                        
#>  collate  en_US.UTF-8                 
#>  ctype    en_US.UTF-8                 
#>  tz       America/New_York            
#>  date     2019-09-20                  
#> 
#> ─ Packages ──────────────────────────────────────────────────────────────
#>  package     * version    date       lib
#>  assertthat    0.2.1      2019-03-21 [1]
#>  backports     1.1.4      2019-04-10 [1]
#>  callr         3.3.1      2019-07-18 [1]
#>  cli           1.1.0      2019-03-19 [1]
#>  crayon        1.3.4      2017-09-16 [1]
#>  desc          1.2.0      2019-07-10 [1]
#>  devtools      2.2.0.9000 2019-09-10 [1]
#>  digest        0.6.20     2019-07-04 [1]
#>  DT            0.8        2019-08-07 [1]
#>  ellipsis      0.2.0.1    2019-07-02 [1]
#>  evaluate      0.14       2019-05-28 [1]
#>  fs            1.3.1      2019-05-06 [1]
#>  glue          1.3.1      2019-03-12 [1]
#>  highr         0.8        2019-03-20 [1]
#>  htmltools     0.3.6      2017-04-28 [1]
#>  htmlwidgets   1.3        2018-09-30 [1]
#>  knitr         1.24.3     2019-08-28 [1]
#>  magrittr      1.5        2014-11-22 [1]
#>  memoise       1.1.0      2017-04-21 [1]
#>  pkgbuild      1.0.5      2019-08-26 [1]
#>  pkgload       1.0.2      2018-10-29 [1]
#>  prettyunits   1.0.2      2015-07-13 [1]
#>  processx      3.4.1      2019-07-18 [1]
#>  ps            1.3.0      2018-12-21 [1]
#>  R6            2.4.0      2019-02-14 [1]
#>  Rcpp          1.0.2      2019-07-25 [1]
#>  read.gt3x   * 0.1.0.9000 2019-09-19 [1]
#>  remotes       2.1.0      2019-06-24 [1]
#>  rlang         0.4.0      2019-06-25 [1]
#>  rmarkdown     1.15       2019-08-21 [1]
#>  rprojroot     1.3-2      2018-01-03 [1]
#>  sessioninfo   1.1.1      2018-11-05 [1]
#>  stringi       1.4.3      2019-03-12 [1]
#>  stringr       1.4.0      2019-02-10 [1]
#>  testthat      2.2.1      2019-07-25 [1]
#>  usethis       1.5.1.9000 2019-08-15 [1]
#>  withr         2.1.2      2018-03-15 [1]
#>  xfun          0.9        2019-08-21 [1]
#>  yaml          2.2.0      2018-07-25 [1]
#>  source                            
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  Github (muschellij2/desc@b0c374f) 
#>  Github (r-lib/devtools@d7f0915)   
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  Github (muschellij2/knitr@abcea3d)
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  local                             
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  local                             
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

TuomoNieminen, 2019-10-06:
Hey,

I look at the time shifted 1's issue. However I did not really find any bugs with the implementation of creating the timestamps for the new GT3X format. It seems to me that this is might be a feature rather than a bug: the first sample is actually saved 1 second after the reported start time.

However I am open to discussion on how this should be handled.

When parsing the old format, is there is an assumption that the first sample is drawn at start time and that there is no pauses in recording?

There was a bug in the ImputeZeroes feature.

fixing commit here: https://github.com/THLfi/read.gt3x/commit/abf225e9950222828c6cfaa0be520474ac6c5568

ImputeZeroes fixed behaviour
----------------------------

``` r
library(read.gt3x)
options(digits = 10, digits.secs = 5)
gt3xfile <- gt3x_datapath(1)
```

    ## Downloading gt3x sample data from https://github.com/THLfi/read.gt3x/releases/download/v1.0/EE_left_29.5.2017-05-30.gt3x.zip

``` r
X_z <- read.gt3x(gt3xfile, imputeZeroes = TRUE)

(ti_z <- attributes(X_z)$time_index)[c(1, length(ti_z))]
```

    ## [1]      100 60684499

``` r
(st <- attributes(X_z)$start_time)
```

    ## [1] "2017-05-23 09:15:00 GMT"

``` r
df_z <- as.data.frame(X_z)
max(df_z$time)
```

    ## [1] "2017-05-30 09:49:04.99 GMT"

``` r
range(df_z$time)
```

    ## [1] "2017-05-23 09:15:01.00 GMT" "2017-05-30 09:49:04.99 GMT"

muschellij2, 2019-10-08:
I think the main issue is that the ActiLife software does not give the sam result.  

``` r
rm(list = ls())
library(read.gt3x)
library(readr)

## Personal dropbox sharing links
csv.fpath <- "https://www.dropbox.com/s/hhdqjntarqxgpuu/TAS1H30182785%20%282019-09-17%29RAW.csv?dl=1"
gt3x.fpath <- "https://www.dropbox.com/s/pi7m9b75gqzl64g/TAS1H30182785%20%282019-09-17%29.gt3x?dl=1"
gt3x.destfile <- tempfile(fileext = '.gt3x')
gt3x.csv <- tempfile(fileext = '.csv.gz')

## Download gt3x to wd
if (!file.exists(gt3x.destfile)) download.file(gt3x.fpath, gt3x.destfile) 
if (!file.exists(gt3x.csv)) download.file(csv.fpath, gt3x.csv) 

actilife = readr::read_csv(gt3x.csv, skip = 10)
#> Parsed with column specification:
#> cols(
#>   `Accelerometer X` = col_double(),
#>   `Accelerometer Y` = col_double(),
#>   `Accelerometer Z` = col_double()
#> )

info = as.character(unlist(read.csv(file = gt3x.csv, nrows = 6)))
info
#> [1] "Serial Number: TAS1H30182785"     "Start Time 18:40:00"             
#> [3] "Start Date 9/17/2019"             "Epoch Period (hh:mm:ss) 00:00:00"
#> [5] "Download Time 19:20:05"           "Download Date 9/17/2019"
actilife = readr::read_csv(gt3x.csv, skip = 10)
#> Parsed with column specification:
#> cols(
#>   `Accelerometer X` = col_double(),
#>   `Accelerometer Y` = col_double(),
#>   `Accelerometer Z` = col_double()
#> )
res = read.gt3x(gt3x.destfile)
head(res)
#> Sampling Rate: 100Hz
#> Firmware Version: 1.7.2
#> Serial Number Prefix: TAS
#>          X      Y     Z
#> [1,] 0.008  0.000 1.016
#> [2,] 0.012 -0.008 1.023
#> [3,] 0.008 -0.008 1.016
#> [4,] 0.012 -0.008 1.012
#> [5,] 0.016 -0.008 1.008
#> [6,] 0.020  0.004 1.008

ag_res = AGread::read_gt3x(gt3x.destfile)$RAW
head(ag_res)
#>             Timestamp Accelerometer_X Accelerometer_Y Accelerometer_Z
#> 1 2019-09-17 18:40:00           0.000           0.008           0.996
#> 2 2019-09-17 18:40:00           0.016           0.000           1.008
#> 3 2019-09-17 18:40:00           0.020          -0.008           1.004
#> 4 2019-09-17 18:40:00           0.016          -0.012           1.012
#> 5 2019-09-17 18:40:00           0.016          -0.008           1.008
#> 6 2019-09-17 18:40:00           0.008          -0.008           1.008
```

<sup>Created on 2019-10-07 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0)</sup>

muschellij2, 2019-10-11:
So I think this illustrates the main problem.  " the first sample is actually saved 1 second after the reported start time." - I don't think that's accurate.  In the example above and below the first lines do not match, this is directly due to the 1s issue.

# Getting the data

Here we will read in the data and the output from ActiLife (gold standard)
which is the CSV:

``` r
library(lubridate)
#> 
#> Attaching package: 'lubridate'
#> The following object is masked from 'package:base':
#> 
#>     date
## Personal dropbox sharing links
csv.fpath <- "https://www.dropbox.com/s/hhdqjntarqxgpuu/TAS1H30182785%20%282019-09-17%29RAW.csv?dl=1"
gt3x.fpath <- "https://www.dropbox.com/s/pi7m9b75gqzl64g/TAS1H30182785%20%282019-09-17%29.gt3x?dl=1"
fname <- tempfile(fileext = '.gt3x')
csv <- tempfile(fileext = '.csv.gz')

## Download gt3x to wd
if (!file.exists(fname)) download.file(gt3x.fpath, fname)
if (!file.exists(csv)) download.file(csv.fpath, csv)
info = readLines(csv, n = 10)
head(info)
#> [1] "------------ Data File Created By ActiGraph GT3X+ ActiLife v6.13.3 Firmware v1.7.2 date format M/d/yyyy at 100 Hz  Filter Normal -----------"
#> [2] "Serial Number: TAS1H30182785"                                                                                                                
#> [3] "Start Time 18:40:00"                                                                                                                         
#> [4] "Start Date 9/17/2019"                                                                                                                        
#> [5] "Epoch Period (hh:mm:ss) 00:00:00"                                                                                                            
#> [6] "Download Time 19:20:05"
correct = readr::read_csv(csv, skip = 10)
#> Parsed with column specification:
#> cols(
#>   `Accelerometer X` = col_double(),
#>   `Accelerometer Y` = col_double(),
#>   `Accelerometer Z` = col_double()
#> )
dim(correct)
#> [1] 240500      3
```

# Reading in the data

## `AGread`

``` r
ag =  AGread::read_gt3x(fname)$RAW
```

## `read.gt3x` without imputing zeroes

``` r
rg = read.gt3x::read.gt3x(fname, asDataFrame = TRUE)
```

## `read.gt3x` with imputing zeroes

``` r
rgi = read.gt3x::read.gt3x(fname, imputeZeroes = TRUE, asDataFrame = TRUE)
```

We see the number of rows vary from all implementations. The `AGread`
implementation agrees with correct and are **identical**.

``` r
dim(ag)
#> [1] 240500      4
dim(rg)
#> [1] 32900     4
[... 305 lines cut ...]
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  local                             
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#>  CRAN (R 3.6.0)                    
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

TuomoNieminen, 2019-10-22:
Hey,

Thanks for that. Could you write small tests which highlight the problmes and show the expected results? I will work on this as soon as I can make time, which at the end of november.

martakarass, 2019-11-17:
Hi! Please find below an example with expected vs actual result. Note this is the same example data as [here](https://github.com/THLfi/read.gt3x/issues/5). Note I used package from its most recent (`1 parent e89c8f1 commit abf225e9950222828c6cfaa0be520474ac6c5568`) commit. 

###  Links to (1) GT3X file, (2) ActiLife raw data output CSV. Download data

```r
rm(list = ls())

library(data.table)
# devtools::install_github("THLfi/read.gt3x", ref = "abf225e9950222828c6cfaa0be520474ac6c5568")
library(read.gt3x)

## Personal dropbox sharing links to (1) GT3X file, (2) ActiLife raw data output CSV
gt3x.fpath <- "https://www.dropbox.com/s/pi7m9b75gqzl64g/TAS1H30182785%20%282019-09-17%29.gt3x?dl=1"
csv.fpath <- "https://www.dropbox.com/s/hhdqjntarqxgpuu/TAS1H30182785%20%282019-09-17%29RAW.csv?dl=1"

file_directory <- getwd()
gt3x.destfile <- file.path(file_directory, "TAS1H30182785 (2019-09-17).gt3x")
csv.destfile <- file.path(file_directory, "TAS1H30182785 (2019-09-17).csv")

## Download files to wd
if (!file.exists(gt3x.destfile)) download.file(gt3x.fpath, gt3x.destfile)
if (!file.exists(csv.destfile)) download.file(csv.fpath, csv.destfile)
```

### Read the data 
```r
## Read ActiLife raw data output CSV
dat1 <- as.data.frame(fread(csv.destfile))
## Read via read.gt3x
dat3 <- read.gt3x(gt3x.destfile, asDataFrame = TRUE, imputeZeroes = TRUE)
```

### Show disagreement with 1s of x-axis values
```r
## First second of ActiLife raw data output CSV: X axis 
dat1$`Accelerometer X`[1:100] 
# [1] 0.000 0.016 0.020 0.016 0.016 0.008 0.016 0.020 0.016 0.012 0.012 0.012 0.008 0.012 0.012 0.016
# [17] 0.012 0.012 0.012 0.016 0.012 0.012 0.012 0.012 0.016 0.016 0.020 0.012 0.012 0.008 0.012 0.020
# [33] 0.020 0.012 0.012 0.012 0.008 0.012 0.016 0.012 0.016 0.008 0.008 0.016 0.020 0.020 0.020 0.012
# [49] 0.012 0.008 0.004 0.016 0.012 0.008 0.008 0.012 0.016 0.016 0.016 0.016 0.012 0.016 0.012 0.016
# [65] 0.020 0.012 0.016 0.016 0.016 0.012 0.016 0.016 0.016 0.012 0.016 0.016 0.008 0.016 0.012 0.008
# [81] 0.008 0.012 0.016 0.016 0.012 0.016 0.016 0.012 0.016 0.012 0.016 0.012 0.016 0.012 0.012 0.008
# [97] 0.008 0.012 0.016 0.020

## First second of read.gt3x read data output: X axis 
dat3$X[1:100]
# [1] 0.008 0.012 0.008 0.012 0.016 0.020 0.012 0.012 0.016 0.016 0.008 0.012 0.012 0.012 0.012 0.008
# [17] 0.016 0.016 0.016 0.016 0.012 0.008 0.012 0.012 0.016 0.012 0.008 0.008 0.016 0.012 0.016 0.020
# [33] 0.012 0.012 0.012 0.012 0.016 0.016 0.020 0.020 0.016 0.012 0.016 0.012 0.012 0.012 0.008 0.008
# [49] 0.008 0.016 0.004 0.016 0.012 0.008 0.008 0.016 0.020 0.016 0.016 0.012 0.012 0.012 0.016 0.020
# [65] 0.012 0.012 0.016 0.016 0.008 0.008 0.012 0.016 0.016 0.012 0.008 0.004 0.008 0.008 0.012 0.012
# [81] 0.012 0.008 0.012 0.016 0.016 0.008 0.008 0.008 0.012 0.012 0.020 0.012 0.012 0.008 0.008 0.008
# [97] 0.012 0.016 0.012 0.016

## See the first second of data from both data sets does not match 
dat1$`Accelerometer X`[1:100] - dat3$X[1:100]
# [1] -0.008  0.004  0.012  0.004  0.000 -0.012  0.004  0.008  0.000 -0.004  0.004  0.000 -0.004  0.000
# [15]  0.000  0.008 -0.004 -0.004 -0.004  0.000  0.000  0.004  0.000  0.000  0.000  0.004  0.012  0.004
# [29] -0.004 -0.004 -0.004  0.000  0.008  0.000  0.000  0.000 -0.008 -0.004 -0.004 -0.008  0.000 -0.004
# [43] -0.008  0.004  0.008  0.008  0.012  0.004  0.004 -0.008  0.000  0.000  0.000  0.000  0.000 -0.004
# [57] -0.004  0.000  0.000  0.004  0.000  0.004 -0.004 -0.004  0.008  0.000  0.000  0.000  0.008  0.004
# [71]  0.004  0.000  0.000  0.000  0.008  0.012  0.000  0.008  0.000 -0.004 -0.004  0.004  0.004  0.000
# [85] -0.004  0.008  0.008  0.004  0.004  0.000 -0.004  0.000  0.004  0.004  0.004  0.000 -0.004 -0.004
# [99]  0.004  0.004
```

### Show agreement when we start looking up the ActiLife raw data output discarding the 1s
```r
all(dat1$`Accelerometer X`[101:200] - dat3$X[1:100] == 0)
# [1] TRUE
[... 64 lines cut ...]
# purrr         0.3.3   2019-10-18 [1] CRAN (R 3.5.2)                  
# R6            2.4.1   2019-11-12 [1] CRAN (R 3.5.2)                  
# Rcpp          1.0.3   2019-11-08 [1] CRAN (R 3.5.2)                  
# read.gt3x   * 0.1.0   2019-11-17 [1] Github (THLfi/read.gt3x@abf225e)
# remotes       2.0.4   2019-04-10 [1] CRAN (R 3.5.2)                  
# reshape2    * 1.4.3   2017-12-11 [1] CRAN (R 3.5.0)                  
# rlang         0.4.1   2019-10-24 [1] CRAN (R 3.5.2)                  
# rprojroot     1.3-2   2018-01-03 [1] CRAN (R 3.5.0)                  
# rstudioapi    0.10    2019-03-19 [1] CRAN (R 3.5.2)                  
# scales        1.0.0   2018-08-09 [1] CRAN (R 3.5.0)                  
# sessioninfo   1.1.1   2018-11-05 [1] CRAN (R 3.5.0)                  
# stringi       1.4.3   2019-03-12 [1] CRAN (R 3.5.2)                  
# stringr       1.4.0   2019-02-10 [1] CRAN (R 3.5.2)                  
# testthat      2.1.1   2019-04-23 [1] CRAN (R 3.5.2)                  
# tibble        2.1.3   2019-06-06 [1] CRAN (R 3.5.2)                  
# tidyselect    0.2.5   2018-10-11 [1] CRAN (R 3.5.0)                  
# usethis       1.5.0   2019-04-07 [1] CRAN (R 3.5.2)                  
# withr         2.1.2   2018-03-15 [1] CRAN (R 3.5.0)                  
# yaml          2.2.0   2018-07-25 [1] CRAN (R 3.5.0)                  
# 
# [1] /Library/Frameworks/R.framework/Versions/3.5/Resources/library
```
</details>

FYI: @muschellij2

TuomoNieminen, 2019-11-30:
I did finally find the cause of the 1s issue. It was a simple bug which caused the parser to skip reading the first payload after parsing the parameters. 

@martakarass Do you mind if I use your data for creating small tests to the package?

martakarass, 2019-11-30:
Hi @TuomoNieminen I do not mind at all! Glad that data can be used to help here.

TuomoNieminen, 2019-12-03:
Fixing the missing 1s issue
================
Tuomo Nieminen
December 3, 2019

Hello,

The missing 1s issue is fixed in this commit: <https://github.com/THLfi/read.gt3x/commit/55344d7c566bb912669caf48f56c4e33e2ef1db4>

I addressed imputation in these commits:

-   Impute after last measurement until device offtime: <https://github.com/THLfi/read.gt3x/commit/7c39bf778c4dffb44b6cab1399b87b45880d1e94>
-   Impute when the raw data has empty measurements: <https://github.com/THLfi/read.gt3x/commit/e9356cfd886ef6eb0abb8c6f53d0ba4bc1320389>

Tests showing correct behaviour are in: tests/testthat/test\_read\_new.R. Some examples as a response to the issue examples are below.

``` r
# Read example data from the /tests directory of read.gt3x source code.
csvfile <- "tests/TAS1H30182785 (2019-09-17).csv"
gt3xfile <- "tests/TAS1H30182785 (2019-09-17).gt3x"

correct <- as.data.frame(data.table::fread(csvfile))

dim(correct)
```

    ## [1] 240500      3

### Data dimensions

``` r
rg = read.gt3x::read.gt3x(gt3xfile, asDataFrame = TRUE)
rgi = read.gt3x::read.gt3x(gt3xfile, imputeZeroes = TRUE, asDataFrame = TRUE)
```

``` r
dim(rg)
```

    ## [1] 33000     4

``` r
dim(rgi)
```

    ## [1] 240500      4

``` r
nrow(rg) == nrow(correct)
```

    ## [1] FALSE

``` r
nrow(rgi) == nrow(correct)
```

    ## [1] TRUE

### The previously missing second

The first second is now included.

``` r
head(correct)
```

    ##   Accelerometer X Accelerometer Y Accelerometer Z
    ## 1           0.000           0.008           0.996
    ## 2           0.016           0.000           1.008
    ## 3           0.020          -0.008           1.004
    ## 4           0.016          -0.012           1.012
    ## 5           0.016          -0.008           1.008
    ## 6           0.008          -0.008           1.008

``` r
head(rg)
```

    ## Sampling Rate: 100Hz
    ## Firmware Version: 1.7.2
    ## Serial Number Prefix: TAS
    ##       X      Y     Z                time
    ## 1 0.000  0.008 0.996 2019-09-17 18:40:00
    ## 2 0.016  0.000 1.008 2019-09-17 18:40:00
    ## 3 0.020 -0.008 1.004 2019-09-17 18:40:00
    ## 4 0.016 -0.012 1.012 2019-09-17 18:40:00
    ## 5 0.016 -0.008 1.008 2019-09-17 18:40:00
    ## 6 0.008 -0.008 1.008 2019-09-17 18:40:00

``` r
head(rgi)
```

    ## Sampling Rate: 100Hz
    ## Firmware Version: 1.7.2
    ## Serial Number Prefix: TAS
    ##       X      Y     Z                time
    ## 1 0.000  0.008 0.996 2019-09-17 18:40:00
    ## 2 0.016  0.000 1.008 2019-09-17 18:40:00
    ## 3 0.020 -0.008 1.004 2019-09-17 18:40:00
    ## 4 0.016 -0.012 1.012 2019-09-17 18:40:00
    ## 5 0.016 -0.008 1.008 2019-09-17 18:40:00
    ## 6 0.008 -0.008 1.008 2019-09-17 18:40:00

### Missing seconds in the middle of data

Imputation now covers cases where the raw data included an activity measurement with no data.

``` r
rgi[which(diff(rgi$time) > 0.1) + 1,]
```

    ## Sampling Rate: 100Hz
    ## Firmware Version: 1.7.2
    ## Serial Number Prefix: TAS
    ## [1] X    Y    Z    time
    ## <0 rows> (or 0-length row.names)

``` r
atrg = attributes(rg)
atrgi = attributes(rgi)
sum(atrg$missingness$n_missing)
```

    ## [1] 207500

``` r
sum(atrgi$missingness$n_missing)
```

    ## [1] 207500

### Recording the missing values as attributes

The numbers of missing values are now correctly recorded.

``` r
nrow(rg) + sum(atrg$missingness$n_missing) == nrow(correct)
```

    ## [1] TRUE

closed 2019-12-03 by TuomoNieminen