THLfi/read.gt3x #6: Memory issues

muschellij2, 2019-09-19:
Should there be a way to chunk the data so you don't have copies in the C++ and R at the same time?  If not, there may be some issues for Windows or large files being read (including the example in the package):

https://builder.r-hub.io/status/read.gt3x_0.1.0.tar.gz-5eef48a06f4d4c0a852f42525780f31f

TuomoNieminen, 2020-01-08:
Hey, first of all apologies not commenting on this earlier.

Could you clarify a little bit on what would chunking the data mean? A request I have received from elsewhere has been to implement a solution to read only part of a file by for example giving the incides or time stamps between which to read the accelerometer data. Is this what you mean?

muschellij2, 2020-01-08:
I believe if you read the entire data format in `C++` and then pass through to `R`, it essentially has 2 copies in memory until the garbage collector frees the `C++` pointer.  I'm not 100% sure of this, but if that is the case, it may be useful to be able to do this in a slower albeit more "memory-saving" way, by reading the data in chunks (let's say of 100000 rows) and then pass to R and iterate.  Not a large issue, but the solution you give by giving a max index or different indices would work for what I need (it to not fail on memory on `rhub` for checking)

closed 2020-11-01 by muschellij2