THLfi/read.gt3x #9: Non-systematic discrepancies in values between ActiLife raw data output CSV and read.gt3x read

martakarass, 2019-11-17:
I have gt3x file with ~40 minutes of raw accelerometry data collected at 100 Hz, with idle mode enabled. I have corresponding ActiLife raw data output CSV. (Note this is the same example data as [here](https://github.com/THLfi/read.gt3x/issues/5).) I try to use `read.gt3x::read.gt3x` to replicate ActiLife raw data output. 

I note there seems to be some non-systematic discrepancies in values between ActiLife raw data output CSV and read.gt3x read. I provide files and code showing the discrepancies with a plot. 

# Links to (1) GT3X file, (2) ActiLife raw data output CSV. Download data

```r
rm(list = ls())

library(data.table)
# devtools::install_github("THLfi/read.gt3x", ref = "abf225e9950222828c6cfaa0be520474ac6c5568")
library(read.gt3x)

## Personal dropbox sharing links to (1) GT3X file, (2) ActiLife raw data output CSV
gt3x.fpath <- "https://www.dropbox.com/s/pi7m9b75gqzl64g/TAS1H30182785%20%282019-09-17%29.gt3x?dl=1"
csv.fpath <- "https://www.dropbox.com/s/hhdqjntarqxgpuu/TAS1H30182785%20%282019-09-17%29RAW.csv?dl=1"

file_directory <- getwd()
gt3x.destfile <- file.path(file_directory, "TAS1H30182785 (2019-09-17).gt3x")
csv.destfile <- file.path(file_directory, "TAS1H30182785 (2019-09-17).csv")

## Download files to wd
if (!file.exists(gt3x.destfile)) download.file(gt3x.fpath, gt3x.destfile)
if (!file.exists(csv.destfile)) download.file(csv.fpath, csv.destfile)
```

# Demonstrate (non-systematic?) discrepancies in x-axis values between ActiLife raw data output CSV and read.gt3x read

Read ActiLife raw data output CSV
```r
dat1 <- as.data.frame(fread(csv.destfile))
```

Read GT3X via read.gt3x
```r
dat3 <- read.gt3x(gt3x.destfile, asDataFrame = TRUE, imputeZeroes = TRUE)
```

Observe mismatch in row number
```
c(nrow(dat1), nrow(dat3))
# [1] 240500 215700
```

Plot values of: 

1. ActiLife raw data output: x-axis,
2. `read.gt3x` data read: x-axis,
3. difference between the two above. 

```
par(mfrow = c(3,1))
plot(dat1$`Accelerometer X`[1:nrow(dat3)], type = "l", col = "red", ylab = "ActiLife raw data output: x-axis")
plot(dat3$X[1:nrow(dat3)], type = "l", col = "blue", ylab = "read.gt3x data read: x-axis")
plot(dat1$`Accelerometer X`[1:nrow(dat3)] - dat3$X[1:nrow(dat3)], type = "l", ylab = "Difference")
```
[image]

<details>
  <summary>Session info</summary>

```r
devtools::session_info()
# ─ Session info ─────────────────────────────────────────────────────────────────────────────────────────
# setting  value                       
# version  R version 3.5.2 (2018-12-20)
# os       macOS Mojave 10.14.2        
# system   x86_64, darwin15.6.0        
# ui       RStudio                     
# language (EN)                        
# collate  en_US.UTF-8                 
# ctype    en_US.UTF-8                 
# tz       America/New_York            
# date     2019-11-17                  
# 
# ─ Packages ─────────────────────────────────────────────────────────────────────────────────────────────
# package     * version date       lib source                          
# assertthat    0.2.1   2019-03-21 [1] CRAN (R 3.5.2)                  
# backports     1.1.5   2019-10-02 [1] CRAN (R 3.5.2)                  
# callr         3.2.0   2019-03-15 [1] CRAN (R 3.5.2)                  
# cli           1.1.0   2019-03-19 [1] CRAN (R 3.5.2)                  
# colorspace    1.4-1   2019-03-18 [1] CRAN (R 3.5.2)                  
# crayon        1.3.4   2017-09-16 [1] CRAN (R 3.5.0)                  
# curl          3.3     2019-01-10 [1] CRAN (R 3.5.2)                  
# data.table  * 1.12.6  2019-10-18 [1] CRAN (R 3.5.2)                  
# desc          1.2.0   2018-05-01 [1] CRAN (R 3.5.0)                  
# devtools      2.0.2   2019-04-08 [1] CRAN (R 3.5.2)                  
# digest        0.6.22  2019-10-21 [1] CRAN (R 3.5.2)                  
# dplyr       * 0.8.3   2019-07-04 [1] CRAN (R 3.5.2)                  
# fs            1.3.1   2019-05-06 [1] CRAN (R 3.5.2)                  
# ggplot2     * 3.2.1   2019-08-10 [1] CRAN (R 3.5.2)                  
# glue          1.3.1   2019-03-12 [1] CRAN (R 3.5.2)                  
# gtable        0.3.0   2019-03-25 [1] CRAN (R 3.5.2)                  
# lazyeval      0.2.2   2019-03-15 [1] CRAN (R 3.5.2)                  
# lubridate   * 1.7.4   2018-04-11 [1] CRAN (R 3.5.0)                  
# magrittr      1.5     2014-11-22 [1] CRAN (R 3.5.0)                  
# memoise       1.1.0   2017-04-21 [1] CRAN (R 3.5.0)                  
# munsell       0.5.0   2018-06-12 [1] CRAN (R 3.5.0)                  
# pillar        1.4.2   2019-06-29 [1] CRAN (R 3.5.2)                  
# pkgbuild      1.0.3   2019-03-20 [1] CRAN (R 3.5.2)                  
# pkgconfig     2.0.3   2019-09-22 [1] CRAN (R 3.5.2)                  
# pkgload       1.0.2   2018-10-29 [1] CRAN (R 3.5.0)                  
# plyr          1.8.4   2016-06-08 [1] CRAN (R 3.5.0)                  
# prettyunits   1.0.2   2015-07-13 [1] CRAN (R 3.5.0)                  
# processx      3.4.1   2019-07-18 [1] CRAN (R 3.5.2)                  
# ps            1.3.0   2018-12-21 [1] CRAN (R 3.5.0)                  
# purrr         0.3.3   2019-10-18 [1] CRAN (R 3.5.2)                  
# R6            2.4.1   2019-11-12 [1] CRAN (R 3.5.2)                  
# Rcpp          1.0.3   2019-11-08 [1] CRAN (R 3.5.2)                  
# read.gt3x   * 0.1.0   2019-11-17 [1] Github (THLfi/read.gt3x@abf225e)
# remotes       2.0.4   2019-04-10 [1] CRAN (R 3.5.2)                  
# reshape2    * 1.4.3   2017-12-11 [1] CRAN (R 3.5.0)                  
# rlang         0.4.1   2019-10-24 [1] CRAN (R 3.5.2)                  
# rprojroot     1.3-2   2018-01-03 [1] CRAN (R 3.5.0)                  
# rstudioapi    0.10    2019-03-19 [1] CRAN (R 3.5.2)                  
# scales        1.0.0   2018-08-09 [1] CRAN (R 3.5.0)                  
# sessioninfo   1.1.1   2018-11-05 [1] CRAN (R 3.5.0)                  
# stringi       1.4.3   2019-03-12 [1] CRAN (R 3.5.2)                  
# stringr       1.4.0   2019-02-10 [1] CRAN (R 3.5.2)                  
# testthat      2.1.1   2019-04-23 [1] CRAN (R 3.5.2)                  
# tibble        2.1.3   2019-06-06 [1] CRAN (R 3.5.2)                  
# tidyselect    0.2.5   2018-10-11 [1] CRAN (R 3.5.0)                  
# usethis       1.5.0   2019-04-07 [1] CRAN (R 3.5.2)                  
# withr         2.1.2   2018-03-15 [1] CRAN (R 3.5.0)                  
# yaml          2.2.0   2018-07-25 [1] CRAN (R 3.5.0)                  
# 
# [1] /Library/Frameworks/R.framework/Versions/3.5/Resources/library
```
</details>

FYI: @muschellij2

TuomoNieminen, 2019-12-03:
Addressing the discrepancies
================
Tuomo Nieminen
December 3, 2019

Hey,

Thanks for this. Most of the discrepancies in the number of rows were due to the fact that the imputation feature did not cover timestamps after the last recorded value. But also a few seconds were missing because there were cases when the data includes an activity measurement for a given second with actual zero rows of data, and these values were not imputed before.

``` r
# Read example data from the /tests directory of read.gt3x source code.
csvfile <- "tests/TAS1H30182785 (2019-09-17).csv"
gt3xfile <- "tests/TAS1H30182785 (2019-09-17).gt3x"

csvdata <- as.data.frame(data.table::fread(csvfile))
gt3xdata <- read.gt3x::read.gt3x(gt3xfile, imputeZeroes = T)
```

### Number of rows

``` r
nrow(csvdata) == nrow(gt3xdata)
```

    ## [1] TRUE

### Values

The imputation strategy of Actilife is different from read.gt3x. They repeat the last known measurement.

``` r
csvdata[150000: 150010, ]
```

    ##        Accelerometer X Accelerometer Y Accelerometer Z
    ## 150000          -1.008          -0.129           0.004
    ## 150001          -1.008          -0.129           0.004
    ## 150002          -1.008          -0.129           0.004
    ## 150003          -1.008          -0.129           0.004
    ## 150004          -1.008          -0.129           0.004
    ## 150005          -1.008          -0.129           0.004
    ## 150006          -1.008          -0.129           0.004
    ## 150007          -1.008          -0.129           0.004
    ## 150008          -1.008          -0.129           0.004
    ## 150009          -1.008          -0.129           0.004
    ## 150010          -1.008          -0.129           0.004

read.gt3x simply imputes zeroes

``` r
gt3xdata[150000: 150010, ]
```

    ##       X Y Z
    ##  [1,] 0 0 0
    ##  [2,] 0 0 0
    ##  [3,] 0 0 0
    ##  [4,] 0 0 0
    ##  [5,] 0 0 0
    ##  [6,] 0 0 0
    ##  [7,] 0 0 0
    ##  [8,] 0 0 0
    ##  [9,] 0 0 0
    ## [10,] 0 0 0
    ## [11,] 0 0 0

Other than the imputed rows, the values in the two data are identical.

``` r
row_imputed <- rowSums(gt3xdata) == 0
values_identical <- gt3xdata[!row_imputed, ] == csvdata[!row_imputed, ]
table(values_identical)
```

    ## values_identical
    ##  TRUE 
    ## 99000

closed 2019-12-03 by TuomoNieminen

martakarass, 2019-12-13:
Thanks for the update! Much appreciated.