THLfi/read.gt3x #10: ENH: Add `imputed` column for indicator if `imputeZeroes=TRUE`

muschellij2, 2019-12-03:
Would it be possible to add an indicator of `imputed = TRUE` when `imputeZeroes = TRUE`?  It is possible to check if all columns are zero, but that may not be sufficient as some rows may have no acceleration but are not non-wear.  

The goal is to do a last observation carried forward so that we can get out exactly what comes out of actiLife software.

TuomoNieminen, 2019-12-03:
Nonwear will not have all zero rows due to gravity.

TuomoNieminen, 2019-12-03:
But yes, that information is also stored in the missingness attribute. There could be an option (for example in the as.data.frame method) or a separate method to add an indicator column?

TuomoNieminen, 2019-12-03:
It is also possible that the imputezeroes feature in the C++ isn't really a good way to handle the imputation task. As long as missingness is recorded correctly while parsing, imputation could be done afterwards, using the incomplete data matrix and the missingness attribute. Therefore there could a separate imputation procedure for example with multiple options.