THLfi/read.gt3x #13: Time is all over the place!?

muschellij2, 2020-03-08:
Is there a stochastic/random process of times?  It seems to stabilize - but I'm unsure as to why the times change!

``` r
library(read.gt3x)
destfile = file.path(tempdir(), "AI12_NEO1F09120034_2017-09-25.gt3x.gz")
if (!file.exists(destfile)) {
  download.file("https://ndownloader.figshare.com/files/21855675",
                destfile = destfile)
}
for (i in 1:10) {
  x = read.gt3x(destfile,
    asDataFrame = TRUE)
  print(range(x$time))
}
#> [1] "2016-10-20 19:32:36 GMT" "2016-11-13 03:32:36 GMT"
#> [1] "2016-04-19 05:23:00 GMT" "2016-05-12 13:23:00 GMT"
#> [1] "2016-04-19 05:23:00 GMT" "2016-05-12 13:23:00 GMT"
#> [1] "2016-04-19 05:23:00 GMT" "2016-05-12 13:23:00 GMT"
#> [1] "2016-04-19 05:23:00 GMT" "2016-05-12 13:23:00 GMT"
#> [1] "2016-04-19 05:23:00 GMT" "2016-05-12 13:23:00 GMT"
#> [1] "2016-04-19 05:23:00 GMT" "2016-05-12 13:23:00 GMT"
#> [1] "2016-04-19 05:23:00 GMT" "2016-05-12 13:23:00 GMT"
#> [1] "2016-04-19 05:23:00 GMT" "2016-05-12 13:23:00 GMT"
#> [1] "2016-04-19 05:23:00 GMT" "2016-05-12 13:23:00 GMT"
```

<sup>Created on 2020-03-08 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0.9001)</sup>

<details style="margin-bottom:10px;">

<summary>Session info</summary>

``` r
sessioninfo::session_info()
#> ─ Session info ───────────────────────────────────────────────────────────────
#>  setting  value                       
#>  version  R version 3.6.2 (2019-12-12)
#>  os       macOS Mojave 10.14.6        
#>  system   x86_64, darwin15.6.0        
#>  ui       X11                         
#>  language (EN)                        
#>  collate  en_US.UTF-8                 
#>  ctype    en_US.UTF-8                 
#>  tz       America/New_York            
#>  date     2020-03-08                  
#> 
#> ─ Packages ───────────────────────────────────────────────────────────────────
#>  package     * version     date       lib source                             
#>  assertthat    0.2.1       2019-03-21 [1] CRAN (R 3.6.0)                     
#>  backports     1.1.5       2019-10-02 [1] CRAN (R 3.6.0)                     
#>  cli           2.0.2       2020-02-28 [1] CRAN (R 3.6.0)                     
#>  crayon        1.3.4       2017-09-16 [1] CRAN (R 3.6.0)                     
#>  digest        0.6.25      2020-02-23 [1] CRAN (R 3.6.0)                     
#>  evaluate      0.14        2019-05-28 [1] CRAN (R 3.6.0)                     
#>  fansi         0.4.1       2020-01-08 [1] CRAN (R 3.6.0)                     
#>  fs            1.3.1       2019-05-06 [1] CRAN (R 3.6.0)                     
#>  glue          1.3.1       2019-03-12 [1] CRAN (R 3.6.0)                     
#>  highr         0.8         2019-03-20 [1] CRAN (R 3.6.0)                     
#>  htmltools     0.4.0       2019-10-04 [1] CRAN (R 3.6.0)                     
#>  knitr         1.28        2020-02-06 [1] CRAN (R 3.6.0)                     
#>  magrittr      1.5         2014-11-22 [1] CRAN (R 3.6.0)                     
#>  pillar        1.4.3       2019-12-20 [1] CRAN (R 3.6.0)                     
#>  pkgconfig     2.0.3       2019-09-22 [1] CRAN (R 3.6.0)                     
#>  purrr         0.3.3       2019-10-18 [1] CRAN (R 3.6.0)                     
#>  R.methodsS3   1.8.0       2020-02-14 [1] CRAN (R 3.6.0)                     
#>  R.oo          1.23.0      2019-11-03 [1] CRAN (R 3.6.0)                     
#>  R.utils       2.9.2-9000  2020-02-29 [1] local                              
#>  Rcpp          1.0.3       2019-11-08 [1] CRAN (R 3.6.2)                     
#>  read.gt3x   * 0.1.0.9000  2020-03-08 [1] local                              
#>  reprex        0.3.0.9001  2020-01-05 [1] Github (tidyverse/reprex@5ae0b29)  
#>  rlang         0.4.5       2020-03-01 [1] CRAN (R 3.6.2)                     
#>  rmarkdown     2.1         2020-01-20 [1] CRAN (R 3.6.0)                     
#>  rstudioapi    0.11.0-9000 2020-02-19 [1] Github (rstudio/rstudioapi@deb9c47)
#>  sessioninfo   1.1.1       2018-11-05 [1] CRAN (R 3.6.0)                     
#>  stringi       1.4.6       2020-02-17 [1] CRAN (R 3.6.0)                     
#>  stringr       1.4.0       2019-02-10 [1] CRAN (R 3.6.0)                     
#>  styler        1.3.2       2020-02-23 [1] CRAN (R 3.6.2)                     
#>  tibble        2.1.3       2019-06-06 [1] CRAN (R 3.6.0)                     
#>  withr         2.1.2       2018-03-15 [1] CRAN (R 3.6.0)                     
#>  xfun          0.12        2020-01-13 [1] CRAN (R 3.6.0)                     
#>  yaml          2.2.1       2020-02-01 [1] CRAN (R 3.6.0)                     
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

muschellij2, 2020-03-08:
``` r
library(read.gt3x)
destfile = file.path(tempdir(), "AI12_NEO1F09120034_2017-09-25.gt3x.gz")
if (!file.exists(destfile)) {
  download.file("https://ndownloader.figshare.com/files/21855675",
                destfile = destfile)
}
for (i in 1:5) {
  x = read.gt3x(destfile, asDataFrame = TRUE)
  print(range(x$time))
  hdr = attributes(x)$header
  print(hdr$`Start Date`)
}
#> [1] "2016-06-25 20:15:58 GMT" "2016-07-19 04:15:58 GMT"
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] "2016-08-13 17:16:58 GMT" "2016-09-06 01:16:57 GMT"
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] "2016-08-13 17:16:58 GMT" "2016-09-06 01:16:57 GMT"
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] "2016-08-13 17:16:58 GMT" "2016-09-06 01:16:57 GMT"
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] "2016-08-13 17:16:58 GMT" "2016-09-06 01:16:57 GMT"
#> [1] "2017-09-25 10:00:00 GMT"
```

<sup>Created on 2020-03-08 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0.9001)</sup>

<details style="margin-bottom:10px;">

<summary>Session info</summary>

``` r
sessioninfo::session_info()
#> ─ Session info ───────────────────────────────────────────────────────────────
#>  setting  value                       
#>  version  R version 3.6.2 (2019-12-12)
#>  os       macOS Mojave 10.14.6        
#>  system   x86_64, darwin15.6.0        
#>  ui       X11                         
#>  language (EN)                        
#>  collate  en_US.UTF-8                 
#>  ctype    en_US.UTF-8                 
#>  tz       America/New_York            
#>  date     2020-03-08                  
#> 
#> ─ Packages ───────────────────────────────────────────────────────────────────
#>  package     * version     date       lib source                             
#>  assertthat    0.2.1       2019-03-21 [1] CRAN (R 3.6.0)                     
#>  backports     1.1.5       2019-10-02 [1] CRAN (R 3.6.0)                     
#>  cli           2.0.2       2020-02-28 [1] CRAN (R 3.6.0)                     
#>  crayon        1.3.4       2017-09-16 [1] CRAN (R 3.6.0)                     
#>  digest        0.6.25      2020-02-23 [1] CRAN (R 3.6.0)                     
#>  evaluate      0.14        2019-05-28 [1] CRAN (R 3.6.0)                     
#>  fansi         0.4.1       2020-01-08 [1] CRAN (R 3.6.0)                     
#>  fs            1.3.1       2019-05-06 [1] CRAN (R 3.6.0)                     
#>  glue          1.3.1       2019-03-12 [1] CRAN (R 3.6.0)                     
#>  highr         0.8         2019-03-20 [1] CRAN (R 3.6.0)                     
#>  htmltools     0.4.0       2019-10-04 [1] CRAN (R 3.6.0)                     
#>  knitr         1.28        2020-02-06 [1] CRAN (R 3.6.0)                     
#>  magrittr      1.5         2014-11-22 [1] CRAN (R 3.6.0)                     
#>  pillar        1.4.3       2019-12-20 [1] CRAN (R 3.6.0)                     
#>  pkgconfig     2.0.3       2019-09-22 [1] CRAN (R 3.6.0)                     
#>  purrr         0.3.3       2019-10-18 [1] CRAN (R 3.6.0)                     
#>  R.methodsS3   1.8.0       2020-02-14 [1] CRAN (R 3.6.0)                     
#>  R.oo          1.23.0      2019-11-03 [1] CRAN (R 3.6.0)                     
#>  R.utils       2.9.2-9000  2020-02-29 [1] local                              
#>  Rcpp          1.0.3       2019-11-08 [1] CRAN (R 3.6.2)                     
#>  read.gt3x   * 0.1.0.9000  2020-03-08 [1] local                              
#>  reprex        0.3.0.9001  2020-01-05 [1] Github (tidyverse/reprex@5ae0b29)  
#>  rlang         0.4.5       2020-03-01 [1] CRAN (R 3.6.2)                     
#>  rmarkdown     2.1         2020-01-20 [1] CRAN (R 3.6.0)                     
#>  rstudioapi    0.11.0-9000 2020-02-19 [1] Github (rstudio/rstudioapi@deb9c47)
#>  sessioninfo   1.1.1       2018-11-05 [1] CRAN (R 3.6.0)                     
#>  stringi       1.4.6       2020-02-17 [1] CRAN (R 3.6.0)                     
#>  stringr       1.4.0       2019-02-10 [1] CRAN (R 3.6.0)                     
#>  styler        1.3.2       2020-02-23 [1] CRAN (R 3.6.2)                     
#>  tibble        2.1.3       2019-06-06 [1] CRAN (R 3.6.0)                     
#>  withr         2.1.2       2018-03-15 [1] CRAN (R 3.6.0)                     
#>  xfun          0.12        2020-01-13 [1] CRAN (R 3.6.0)                     
#>  yaml          2.2.1       2020-02-01 [1] CRAN (R 3.6.0)                     
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

vincentvanhees, 2020-03-12:
I had trouble processing your file, which may relate to how OSx zips files (I am working in Ubuntu). When I try it with the example file that comes with [read.gt3x repo](https://github.com/THLfi/read.gt3x/releases) (EE_left_29.5.2017-05-30.gt3x) it all works fine for me:

```
library(read.gt3x)
destfile <- gt3x_datapath(1)
for (i in 1:5) {
  print(paste0("iteration ",i))
  x = read.gt3x(destfile, asDataFrame = T)
  print(range(x$time))
  hdr = attributes(x)$header
  print(hdr$`Start Date`)
  if (i == 5) {
    print("parse_gt3x_info(destfile)")
    print(parse_gt3x_info(destfile))
    print("str(x)")
    print(str(x))
    print("attributes(x)")
    print(names(attributes(x)))
  }
  rm(x)
}
```
**Which results in:**
```
[1] "iteration 1"
[1] "2017-05-23 09:15:00 GMT" "2017-05-30 09:49:03 GMT"
NULL
[1] "iteration 2"
[1] "2017-05-23 09:15:00 GMT" "2017-05-30 09:49:03 GMT"
NULL
[1] "iteration 3"
[1] "2017-05-23 09:15:00 GMT" "2017-05-30 09:49:03 GMT"
NULL
[1] "iteration 4"
[1] "2017-05-23 09:15:00 GMT" "2017-05-30 09:49:03 GMT"
NULL
[1] "iteration 5"
[1] "2017-05-23 09:15:00 GMT" "2017-05-30 09:49:03 GMT"
NULL
[1] "parse_gt3x_info(destfile)"
GT3X information
 $ Serial Number     :"TAS1F27160060"
 $ Device Type       :"Link"
 $ Firmware          :"1.7.0"
 $ Battery Voltage   :"3,86"
 $ Sample Rate       :100
 $ Start Date        : POSIXct, format: "2017-05-23 09:15:00"
 $ Stop Date         : POSIXct, format: "1-01-01"
 $ Last Sample Time  : POSIXct, format: "2017-05-30 09:50:29"
 $ TimeZone          :"03:00:00"
 $ Download Date     : POSIXct, format: "2017-05-30 09:50:29"
 $ Board Revision    :"6"
 $ Unexpected Resets :"0"
 $ Acceleration Scale:256
 $ Acceleration Min  :"-8.0"
 $ Acceleration Max  :"8.0"
 $ Limb              :"Wrist"
 $ Dominance         :"Non-Dominant"
 $ Subject Name      :"EE_left_29.5."
 $ Serial Prefix     :"TAS"
[1] "str(x)"
Classes ‘activity_df’ and 'data.frame':	60684200 obs. of  4 variables:
 $ X   : num  -0.035 -0.008 0.008 0.008 0.004 0 0 -0.004 -0.02 -0.027 ...
 $ Y   : num  0.348 0.352 0.332 0.32 0.316 0.301 0.297 0.316 0.336 0.359 ...
 $ Z   : num  -0.965 -0.934 -0.887 -0.848 -0.844 -0.852 -0.883 -0.918 -0.938 -0.941 ...
 $ time: POSIXct, format: "2017-05-23 09:15:00" "2017-05-23 09:15:00" "2017-05-23 09:15:00" "2017-05-23 09:15:00" ...
 - attr(*, "subject_name")= chr "EE_left_29.5."
 - attr(*, "time_zone")= chr "03:00:00"
 - attr(*, "missingness")='data.frame':	5 obs. of  2 variables:
  ..$ time     : POSIXct, format: "2017-05-23 14:17:40" "2017-05-28 10:06:35" "2017-05-30 09:49:04" "2017-05-30 09:49:05" ...
  ..$ n_missing: int  100 100 100 100 8300
 - attr(*, "old_version")= logi FALSE
 - attr(*, "firmware")= chr "1.7.0"
 - attr(*, "serial_prefix")= chr "TAS"
 - attr(*, "sample_rate")= int 100
NULL
[1] "attributes(x)"
 [1] "names"         "row.names"     "class"         "subject_name"  "time_zone"     "missingness"   "old_version"   "firmware"     
 [9] "serial_prefix" "sample_rate"  
```

**Observations:**
'Start Date' could not be retrieved from object x from this file, because it is not there for this particular file.

**Session info:**
```
> sessionInfo()
R version 3.6.3 (2020-02-29)
Platform: x86_64-pc-linux-gnu (64-bit)
Running under: Ubuntu 18.04.4 LTS

Matrix products: default
BLAS:   /usr/lib/x86_64-linux-gnu/blas/libblas.so.3.7.1
LAPACK: /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3.7.1

locale:
 [1] LC_CTYPE=en_US.UTF-8       LC_NUMERIC=C               LC_TIME=nl_NL.UTF-8        LC_COLLATE=en_US.UTF-8    
 [5] LC_MONETARY=nl_NL.UTF-8    LC_MESSAGES=en_US.UTF-8    LC_PAPER=nl_NL.UTF-8       LC_NAME=C                 
 [9] LC_ADDRESS=C               LC_TELEPHONE=C             LC_MEASUREMENT=nl_NL.UTF-8 LC_IDENTIFICATION=C       

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] read.gt3x_0.1.0

loaded via a namespace (and not attached):
[1] compiler_3.6.3 tools_3.6.3    Rcpp_1.0.3     packrat_0.5.0 
```

muschellij2, 2020-03-13:
Well Can you download the file I use and unzip it and try it?  I think it may be due to the older NHANES format.

vincentvanhees, 2020-03-13:
I gave up too quickly. I have now managed to get it to run with you file (AI12_NEO1F09120034_2017-09-25.gt3x):

Attempt 1:
```
[1] "2018-01-14 19:22:34 GMT" "2018-02-07 03:22:34 GMT"
NULL
[1] "2018-01-22 09:36:45 GMT" "2018-02-14 17:36:45 GMT"
NULL
[1] "2018-01-22 09:36:45 GMT" "2018-02-14 17:36:45 GMT"
NULL
[1] "2018-01-22 09:36:45 GMT" "2018-02-14 17:36:45 GMT"
NULL
[1] "2018-01-22 09:36:45 GMT" "2018-02-14 17:36:45 GMT"
NULL
```

The first timestamp differs. To see whether I can reproduce this I re-ran the code without restarting R, which gives:

```
[1] "iteration 1"
[1] "2018-01-22 09:36:45 GMT" "2018-02-14 17:36:45 GMT"
NULL
[1] "iteration 2"
[1] "2018-01-22 09:36:45 GMT" "2018-02-14 17:36:45 GMT"
NULL
[1] "iteration 3"
[1] "2018-01-22 09:36:45 GMT" "2018-02-14 17:36:45 GMT"
NULL
[1] "iteration 4"
[1] "2018-01-22 09:36:45 GMT" "2018-02-14 17:36:45 GMT"
NULL
[1] "iteration 5"
[1] "2018-01-22 09:36:45 GMT" "2018-02-14 17:36:45 GMT"
NULL
```

So, I agree something is going wrong the first time.

On a side note: Start Date is not correctly extracted, because the header is not extracted by read.gt3x. I see that the header is available in the info file and can inspect it with parse_gt3x_info:
```
 "parse_gt3x_info(destfile)"
GT3X information
 $ Serial Number     :"NEO1F09120034"
 $ Device Type       :"GT3XPlus"
 $ Firmware          :"3.2.1"
 $ Battery Voltage   :"3.98"
 $ Sample Rate       :30
 $ Start Date        : POSIXct, format: "2017-09-25 10:00:00"
 $ Stop Date         : POSIXct, format: "2017-10-02 10:00:00"
 $ Last Sample Time  : POSIXct, format: "2017-10-02 10:00:00"
 $ TimeZone          :"01:00:00"
 $ Download Date     : POSIXct, format: "2017-10-03 10:25:12"
 $ Board Revision    :"5"
 $ Unexpected Resets :"0"
 $ Acceleration Scale:341
 $ Acceleration Min  :"-6.0"
 $ Acceleration Max  :"6.0"
 $ Sex               :"Male"
 $ Limb              :"Wrist"
 $ Side              :"Right"
 $ Dominance         :"Dominant"
 $ Subject Name      :"H11"
 $ Serial Prefix     :"NEO"
```

muschellij2, 2020-03-13:
That's what I've seen.  That's a pretty concerning bug and I'm not sure how
or where it's introduced in the code (first vs. second run).
Best,
John

vincentvanhees, 2020-03-13:
Other issue: Recording year in header does not correspond to year in timestamps.

muschellij2, 2020-03-13:
Ahh - yes I see what you were intending. Agreed.
Best,
John

muschellij2, 2020-03-16:
I looked into this a bit more and found it was the `time_index` that seemed to be reading incorrectly:

``` r
library(read.gt3x)
destfile = file.path(tempdir(), "AI12_NEO1F09120034_2017-09-25.gt3x.gz")
if (!file.exists(destfile)) {
  download.file("https://ndownloader.figshare.com/files/21855675",
                destfile = destfile)
}
for (i in 1:5) {
  acc = read.gt3x::read.gt3x(destfile)
  at = attributes(acc)
  print(at$start_time)
  print(head(at$time_index))
}
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] 832091680 832091683 832091687 832091690 832091693 832091697
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] 309504640 309504643 309504647 309504650 309504653 309504657
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] 309504640 309504643 309504647 309504650 309504653 309504657
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] 309504640 309504643 309504647 309504650 309504653 309504657
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] 309504640 309504643 309504647 309504650 309504653 309504657
```

<sup>Created on 2020-03-16 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0.9001)</sup>

<details style="margin-bottom:10px;">

<summary>Session info</summary>

``` r
sessioninfo::session_info()
#> ─ Session info ───────────────────────────────────────────────────────────────
#>  setting  value                       
#>  version  R version 3.6.2 (2019-12-12)
#>  os       macOS Mojave 10.14.6        
#>  system   x86_64, darwin15.6.0        
#>  ui       X11                         
#>  language (EN)                        
#>  collate  en_US.UTF-8                 
#>  ctype    en_US.UTF-8                 
#>  tz       America/New_York            
#>  date     2020-03-16                  
#> 
#> ─ Packages ───────────────────────────────────────────────────────────────────
#>  package     * version     date       lib source                             
#>  assertthat    0.2.1       2019-03-21 [1] CRAN (R 3.6.0)                     
#>  backports     1.1.5       2019-10-02 [1] CRAN (R 3.6.0)                     
#>  cli           2.0.2       2020-02-28 [1] CRAN (R 3.6.0)                     
#>  crayon        1.3.4       2017-09-16 [1] CRAN (R 3.6.0)                     
#>  digest        0.6.25      2020-02-23 [1] CRAN (R 3.6.0)                     
#>  evaluate      0.14        2019-05-28 [1] CRAN (R 3.6.0)                     
#>  fansi         0.4.1       2020-01-08 [1] CRAN (R 3.6.0)                     
#>  fs            1.3.1       2019-05-06 [1] CRAN (R 3.6.0)                     
#>  glue          1.3.1       2019-03-12 [1] CRAN (R 3.6.0)                     
#>  highr         0.8         2019-03-20 [1] CRAN (R 3.6.0)                     
#>  htmltools     0.4.0       2019-10-04 [1] CRAN (R 3.6.0)                     
#>  knitr         1.28        2020-02-06 [1] CRAN (R 3.6.0)                     
#>  magrittr      1.5         2014-11-22 [1] CRAN (R 3.6.0)                     
#>  pillar        1.4.3       2019-12-20 [1] CRAN (R 3.6.0)                     
#>  pkgconfig     2.0.3       2019-09-22 [1] CRAN (R 3.6.0)                     
#>  purrr         0.3.3       2019-10-18 [1] CRAN (R 3.6.0)                     
#>  R.methodsS3   1.8.0       2020-02-14 [1] CRAN (R 3.6.0)                     
#>  R.oo          1.23.0      2019-11-03 [1] CRAN (R 3.6.0)                     
#>  R.utils       2.9.2-9000  2020-02-29 [1] local                              
#>  Rcpp          1.0.3       2019-11-08 [1] CRAN (R 3.6.2)                     
#>  read.gt3x   * 0.1.0.9000  2020-03-16 [1] local                              
#>  reprex        0.3.0.9001  2020-01-05 [1] Github (tidyverse/reprex@5ae0b29)  
#>  rlang         0.4.5       2020-03-01 [1] CRAN (R 3.6.2)                     
#>  rmarkdown     2.1         2020-01-20 [1] CRAN (R 3.6.0)                     
#>  rstudioapi    0.11.0-9000 2020-02-19 [1] Github (rstudio/rstudioapi@deb9c47)
#>  sessioninfo   1.1.1       2018-11-05 [1] CRAN (R 3.6.0)                     
#>  stringi       1.4.6       2020-02-17 [1] CRAN (R 3.6.0)                     
#>  stringr       1.4.0       2019-02-10 [1] CRAN (R 3.6.0)                     
#>  styler        1.3.2       2020-02-23 [1] CRAN (R 3.6.2)                     
#>  tibble        2.1.3       2019-06-06 [1] CRAN (R 3.6.0)                     
#>  withr         2.1.2       2018-03-15 [1] CRAN (R 3.6.0)                     
#>  xfun          0.12        2020-01-13 [1] CRAN (R 3.6.0)                     
#>  yaml          2.2.1       2020-02-01 [1] CRAN (R 3.6.0)                     
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

muschellij2, 2020-03-16:
There is also something odd with the missingness 
``` r
library(read.gt3x)
destfile = file.path(tempdir(), "AI12_NEO1F09120034_2017-09-25.gt3x.gz")
if (!file.exists(destfile)) {
  download.file("https://ndownloader.figshare.com/files/21855675",
                destfile = destfile)
}
for (i in 1:3) {
  acc = read.gt3x::read.gt3x(destfile)
  at = attributes(acc)
  print("Missing data")
  print(head(at$missingness))
  print("N missing")
  print(sum(at$missingness$n_missing))
  dtime = difftime(at$stop_time, time2 = at$start_time, units = "secs")
  print("Range of time minus the data given")
  print(as.numeric(dtime * at$sample_rate) - nrow(acc))
  print(at$start_time)
  print("Time indices")
  print(c(head(at$time_index), tail(at$time_index)))
  df = as.data.frame(acc)
  print(c(head(df$time), tail(df$time)))  
}
#> [1] "Missing data"
#>                           time   n_missing
#> 32766      1970-01-01 09:06:06 -2055615236
#> 1506333777 2017-09-25 10:02:57         150
#> 1506333968 2017-09-25 10:06:08          90
#> 1506334040 2017-09-25 10:07:20         540
#> 1506334133 2017-09-25 10:08:53         300
#> 1506334154 2017-09-25 10:09:14         510
#> [1] "N missing"
#> [1] 7352700
#> [1] "Range of time minus the data given"
#> [1] 7352700
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] "Time indices"
#>  [1] -238388128 -238388125 -238388121 -238388118 -238388115 -238388111
#>  [7] -177908148 -177908145 -177908141 -177908138 -177908135 -177908131
#>  [1] "2017-06-25 06:42:09 EDT" "2017-06-25 06:42:09 EDT"
#>  [3] "2017-06-25 06:42:09 EDT" "2017-06-25 06:42:09 EDT"
#>  [5] "2017-06-25 06:42:09 EDT" "2017-06-25 06:42:09 EDT"
#>  [7] "2017-07-18 14:42:08 EDT" "2017-07-18 14:42:08 EDT"
#>  [9] "2017-07-18 14:42:08 EDT" "2017-07-18 14:42:08 EDT"
#> [11] "2017-07-18 14:42:08 EDT" "2017-07-18 14:42:08 EDT"
#> [1] "Missing data"
#>                           time   n_missing
#> 1          1970-01-01 00:00:01 -2054632286
#> 1506333777 2017-09-25 10:02:57         150
#> 1506333968 2017-09-25 10:06:08          90
#> 1506334040 2017-09-25 10:07:20         540
#> 1506334133 2017-09-25 10:08:53         300
#> 1506334154 2017-09-25 10:09:14         510
#> [1] "N missing"
#> [1] 7352700
#> [1] "Range of time minus the data given"
#> [1] 7352700
#> [1] "2017-09-25 10:00:00 GMT"
#> [1] "Time indices"
#>  [1] -2000975764 -2000975761 -2000975757 -2000975754 -2000975751 -2000975747
#>  [7] -1940495784 -1940495781 -1940495777 -1940495774 -1940495771 -1940495767
#>  [1] "2015-08-15 06:26:47 EDT" "2015-08-15 06:26:47 EDT"
#>  [3] "2015-08-15 06:26:48 EDT" "2015-08-15 06:26:48 EDT"
#>  [5] "2015-08-15 06:26:48 EDT" "2015-08-15 06:26:48 EDT"
#>  [7] "2015-09-07 14:26:47 EDT" "2015-09-07 14:26:47 EDT"
#>  [9] "2015-09-07 14:26:47 EDT" "2015-09-07 14:26:47 EDT"
#> [11] "2015-09-07 14:26:47 EDT" "2015-09-07 14:26:47 EDT"
#> [1] "Missing data"
#>                           time   n_missing
[... 57 lines cut ...]
#>  pillar        1.4.3       2019-12-20 [1] CRAN (R 3.6.0)                     
#>  pkgconfig     2.0.3       2019-09-22 [1] CRAN (R 3.6.0)                     
#>  purrr         0.3.3       2019-10-18 [1] CRAN (R 3.6.0)                     
#>  R.methodsS3   1.8.0       2020-02-14 [1] CRAN (R 3.6.0)                     
#>  R.oo          1.23.0      2019-11-03 [1] CRAN (R 3.6.0)                     
#>  R.utils       2.9.2-9000  2020-02-29 [1] local                              
#>  Rcpp          1.0.3       2019-11-08 [1] CRAN (R 3.6.2)                     
#>  read.gt3x   * 0.1.0.9000  2020-03-16 [1] local                              
#>  reprex        0.3.0.9001  2020-01-05 [1] Github (tidyverse/reprex@5ae0b29)  
#>  rlang         0.4.5       2020-03-01 [1] CRAN (R 3.6.2)                     
#>  rmarkdown     2.1         2020-01-20 [1] CRAN (R 3.6.0)                     
#>  rstudioapi    0.11.0-9000 2020-02-19 [1] Github (rstudio/rstudioapi@deb9c47)
#>  sessioninfo   1.1.1       2018-11-05 [1] CRAN (R 3.6.0)                     
#>  stringi       1.4.6       2020-02-17 [1] CRAN (R 3.6.0)                     
#>  stringr       1.4.0       2019-02-10 [1] CRAN (R 3.6.0)                     
#>  styler        1.3.2       2020-02-23 [1] CRAN (R 3.6.2)                     
#>  tibble        2.1.3       2019-06-06 [1] CRAN (R 3.6.0)                     
#>  withr         2.1.2       2018-03-15 [1] CRAN (R 3.6.0)                     
#>  xfun          0.12        2020-01-13 [1] CRAN (R 3.6.0)                     
#>  yaml          2.2.1       2020-02-01 [1] CRAN (R 3.6.0)                     
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

muschellij2, 2020-03-16:
In summary:
1.  The `time_index` seems wrong - this is how the time is created - so root cause of this issue.
2.  This seems tied to the `missingness` vector.
3.  This file segfaults to impute zeroes.

muschellij2, 2020-03-27:
Here is the associated CSV that should be outputted (from https://springernature.figshare.com/articles/1s_epoch_activity_count_data_stored_as_csv/7946168) so that we can make a clearer comparison for future discussion in this thread
[AI12 NEO1F09120034 (2017-09-25)1secDataTable.csv.zip](https://github.com/THLfi/read.gt3x/files/4395140/AI12.NEO1F09120034.2017-09-25.1secDataTable.csv.zip)

``` r
library(readr)
library(lubridate)
#> 
#> Attaching package: 'lubridate'
#> The following object is masked from 'package:base':
#> 
#>     date
library(read.gt3x)
destfile = file.path(tempdir(), "AI12_NEO1F09120034_2017-09-25.gt3x.gz")
if (!file.exists(destfile)) {
  download.file("https://ndownloader.figshare.com/files/21855675",
                destfile = destfile)
}
destfile = R.utils::gunzip(destfile, overwrite = TRUE)
acc = read.gt3x::read.gt3x(destfile, asDataFrame = TRUE)
head(acc)
#> Sampling Rate: 30Hz
#> Firmware Version: 3.2.1
#> Serial Number Prefix: NEO
#>        X     Y      Z                time
#> 1 -0.032 0.328 -0.959 2018-01-21 13:26:41
#> 2 -0.026 0.334 -0.950 2018-01-21 13:26:41
#> 3 -0.026 0.331 -0.950 2018-01-21 13:26:41
#> 4 -0.021 0.337 -0.947 2018-01-21 13:26:41
#> 5 -0.029 0.343 -0.950 2018-01-21 13:26:41
#> 6 -0.021 0.314 -0.959 2018-01-21 13:26:41
at = attributes(acc)
  
fname = "AI12 NEO1F09120034 (2017-09-25)1secDataTable.csv"
outfile = file.path(tempdir(), fname)
if (!file.exists(outfile)) {
  zip = paste0(
    "https://github.com/THLfi/read.gt3x/", 
    "files/4395140/AI12.NEO1F09120034.2017-09-25.1secDataTable.csv.zip")
  destfile = tempfile(fileext = ".zip")
  download.file(zip, destfile = destfile)
  unz = unzip(destfile, exdir = tempdir())
}
truth = read_csv(
  outfile,
  skip = 10)
#> Parsed with column specification:
#> cols(
#>   Date = col_character(),
#>   Time = col_time(format = ""),
#>   Axis1 = col_double(),
#>   Axis2 = col_double(),
#>   Axis3 = col_double(),
#>   Steps = col_double(),
#>   Lux = col_double(),
#>   `Inclinometer Off` = col_double(),
#>   `Inclinometer Standing` = col_double(),
#>   `Inclinometer Sitting` = col_double(),
#>   `Inclinometer Lying` = col_double(),
#>   `Vector Magnitude` = col_double()
#> )
truth$Date = dmy_hms(paste0(truth$Date, " ", truth$Time))
head(truth)
#> # A tibble: 6 x 12
#>   Date                Time     Axis1 Axis2 Axis3 Steps   Lux `Inclinometer O…
#>   <dttm>              <time>   <dbl> <dbl> <dbl> <dbl> <dbl>            <dbl>
#> 1 2017-09-25 10:00:00 10:00:00     0     0     0     0     0                1
#> 2 2017-09-25 10:00:01 10:00:01     0     0     0     0     0                1
#> 3 2017-09-25 10:00:02 10:00:02     0     0     0     0     0                1
#> 4 2017-09-25 10:00:03 10:00:03     0     0     0     0     0                1
#> 5 2017-09-25 10:00:04 10:00:04     0     0     0     0     0                1
[... 57 lines cut ...]
#>  R.oo          1.23.0      2019-11-03 [1] CRAN (R 3.6.0)                     
#>  R.utils       2.9.2-9000  2020-02-29 [1] local                              
#>  R6            2.4.1       2019-11-12 [1] CRAN (R 3.6.0)                     
#>  Rcpp          1.0.4.4     2020-03-23 [1] Github (RcppCore/Rcpp@9bf01e7)     
#>  read.gt3x   * 0.1.0.9000  2020-03-17 [1] local                              
#>  readr       * 1.3.1       2018-12-21 [1] CRAN (R 3.6.0)                     
#>  reprex        0.3.0.9001  2020-01-05 [1] Github (tidyverse/reprex@5ae0b29)  
#>  rlang         0.4.5.9000  2020-03-24 [1] Github (r-lib/rlang@a90b04b)       
#>  rmarkdown     2.1         2020-01-20 [1] CRAN (R 3.6.0)                     
#>  rstudioapi    0.11.0-9000 2020-02-19 [1] Github (rstudio/rstudioapi@deb9c47)
#>  sessioninfo   1.1.1       2018-11-05 [1] CRAN (R 3.6.0)                     
#>  stringi       1.4.6       2020-02-17 [1] CRAN (R 3.6.0)                     
#>  stringr       1.4.0       2019-02-10 [1] CRAN (R 3.6.0)                     
#>  styler        1.3.2       2020-02-23 [1] CRAN (R 3.6.2)                     
#>  tibble        2.1.3       2019-06-06 [1] CRAN (R 3.6.0)                     
#>  utf8          1.1.4       2018-05-24 [1] CRAN (R 3.6.0)                     
#>  vctrs         0.2.99.9010 2020-03-24 [1] Github (r-lib/vctrs@71aedcf)       
#>  withr         2.1.2       2018-03-15 [1] CRAN (R 3.6.0)                     
#>  xfun          0.12        2020-01-13 [1] CRAN (R 3.6.0)                     
#>  yaml          2.2.1       2020-02-01 [1] CRAN (R 3.6.0)                     
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

TuomoNieminen, 2020-05-03:
Date parsing with 30Hz
================
Tuomo Nieminen
May 3, 2020

In addition to problems related to the start time not being available from the binary file (fixed, see #15), there was a bug in parsing datetimes of non-100Hz samples.

Background
read.gt3x currently uses a a concept of *time unit* as well as *sample rate* when saving and parsing times. All time indeces are originally stored as fractions (sample rate) of 100th of a second (time unit), relative to the starting time (time 0). The as.data.frame method then converts the indices to actual datetime.

The as.data.frame function had a bug where the sample rate, instead of the time unit, was used. The bug did not affect cases where sample rate = 100 as that matched the time unit.

This is now fixed. I also make a mental note that the concept of time unit is probably unnecessary and the implementation of datetime parsing should perhaps be redesigned.

``` r
library(readr)
library(lubridate)
library(read.gt3x)

# read.gt3x
destfile = file.path(tempdir(), "AI12_NEO1F09120034_2017-09-25.gt3x.gz")
if (!file.exists(destfile)) {
  download.file("https://ndownloader.figshare.com/files/21855675",
                destfile = destfile)
}
destfile = R.utils::gunzip(destfile, overwrite = TRUE)

# reference
fname = "AI12 NEO1F09120034 (2017-09-25)1secDataTable.csv"
dir <- tempdir()
outfile = file.path(dir, fname)
if (!file.exists(outfile)) {
  zip = paste0(
    "https://github.com/THLfi/read.gt3x/", 
    "files/4395140/AI12.NEO1F09120034.2017-09-25.1secDataTable.csv.zip")
  destfile2 = tempfile(fileext = ".zip")
  download.file(zip, destfile = destfile2)
  unz = unzip(destfile2, exdir = dir)
}
truth = read_csv(
  outfile,
  skip = 10)
```

Comparison

``` r
acc = read.gt3x::read.gt3x(destfile, asDataFrame = TRUE, imputeZeroes = TRUE)
info <- parse_gt3x_info(destfile)

truth$Date = dmy_hms(paste0(truth$Date, " ", truth$Time))

info$`Start Date`
```

    ## [1] "2017-09-25 10:00:00 GMT"

``` r
info$`Stop Date`
```

    ## [1] "2017-10-02 10:00:00 GMT"

``` r
range(acc$time)
```

    ## [1] "2017-09-25 10:00:00 GMT" "2017-10-02 09:59:59 GMT"

``` r
range(truth$Date)
```

    ## [1] "2017-09-25 10:00:00 UTC" "2017-10-02 09:59:59 UTC"

muschellij2, 2020-05-05:
This is indeed fixed and referenced in #15 - Thanks @TuomoNieminen - that's amazing work

closed 2020-05-05 by muschellij2