THLfi/read.gt3x #16: Potential mix between Y and X values.

angerhang, 2020-04-22:
I am implementing a java version of GT3X, and am comparing my results with yours. Our `activity2` results seem to be consistent but our `activity1` is different. I suspect this is due to that you didn't flip your Y and X-axis when reading `activity1`? 

Do correct me if I am wrong as I didn't dig into the code in great detail. 
https://github.com/THLfi/read.gt3x/blob/2a5683f534dab1d3761d9cb7f9f5da4e55c2ab27/src/parseGT3X.cpp#L223

muschellij2, 2020-05-05:
(Not my repo, but) I would highly recommend if you have a question, please provide a MCVE: https://stackoverflow.com/help/mcve.  In any example, I recommend using a reproducible example using the reprex package (https://github.com/tidyverse/reprex).  Also, please include a `sessioninfo::session_info()` output.

muschellij2, 2020-11-01:
Please see #21 and the fixed that has been presented there.

closed 2020-11-01 by muschellij2