THLfi/read.gt3x #21: ACTIVITY YXZ

muschellij2, 2020-09-01:
I can't tell if ACTIVITY packets are YXZ ordered in the parseActivity function because I think `j` simply puts in `XYZ` order: https://github.com/THLfi/read.gt3x/blob/master/src/parseGT3X.cpp#L223
as per https://github.com/actigraph/GT3X-File-Format/blob/master/LogRecords/Activity.md#description

muschellij2, 2020-09-01:
In the example they provide, they reorder the output here
```
            new AccelerationSample(sample[1], sample[0], sample[2],
                    FirstSample.AddTicks(milliSeconds*TimeSpan.TicksPerMillisecond));
```

muschellij2, 2020-09-04:
OK - I don't know if `ACTIVITY` and `ACTIVITY2` packets can *both* be in there, but I haven't found an example that.  I just added some code to fix that at 

Here's a `reprex` to show you that we need to switch `X` and `Y`:

``` r
library(read.gt3x)
library(readr)
library(lubridate)
#> 
#> Attaching package: 'lubridate'
#> The following objects are masked from 'package:base':
#> 
#>     date, intersect, setdiff, union
```

# Where is Data from?

The Figshare project for this data is located at https://springernature.figshare.com/collections/Upper\_limb\_activity\_of\_twenty\_myoelectric\_prosthesis\_users\_and\_twenty\_healthy\_anatomically\_intact\_adults\_/4457855. The gt3x data is located at https://springernature.figshare.com/articles/Unprocessed\_raw\_30Hz\_acceleration\_data\_stored\_as\_gt3x/7946189. This data consists of one directory of GT3X files. In order to download them, you need to download the whole data set. As the data are licensed under \[CC0\](https://creativecommons.org/publicdomain/zero/1.0/, we have copied the to another Figshare location (https://figshare.com/articles/dataset/Raw\_GT3X\_Data\_Duplicate/11916087), where each `gt3x` file has been renamed without spaces or parentheses, each file can be downloaded separately, and each `gt3x` file has been compressed for faster transfer and smaller storage footprint.

Also, the raw CSVs from the ActiLife output is located at https://springernature.figshare.com/articles/dataset/Unprocessed\_raw\_30Hz\_acceleration\_data\_stored\_as\_csv/7946186, which we have separated out into individual files (that line up with the GT3X files) at: https://figshare.com/articles/dataset/RAW\_CSV\_from\_GT3X\_Duplicate/12883463.

``` r
idf = list(name_gt3x = "PU3_CLE2B21130054_2017-03-16.gt3x.gz", 
       download_url_gt3x = "https://ndownloader.figshare.com/files/21855807", 
       id = "PU3", serial = "CLE2B21130054", name_csv = "PU3_CLE2B21130054_2017-03-16.csv.gz", 
       download_url_csv = "https://ndownloader.figshare.com/files/24488492")
print(idf)
#> $name_gt3x
#> [1] "PU3_CLE2B21130054_2017-03-16.gt3x.gz"
#> 
#> $download_url_gt3x
#> [1] "https://ndownloader.figshare.com/files/21855807"
#> 
#> $id
#> [1] "PU3"
#> 
#> $serial
#> [1] "CLE2B21130054"
#> 
#> $name_csv
#> [1] "PU3_CLE2B21130054_2017-03-16.csv.gz"
#> 
#> $download_url_csv
#> [1] "https://ndownloader.figshare.com/files/24488492"
download = function(url, name) {
  destfile = file.path(tempdir(), name)
  if (!file.exists(destfile)) {
    download.file(url, destfile)
  }
  destfile
}
gt3x_file = download(idf$download_url_gt3x, idf$name_gt3x)
act_df = read.gt3x(gt3x_file, verbose = TRUE,
                   asDataFrame = TRUE, imputeZeroes = TRUE)
#> Input is a .gt3x file, unzipping to a temporary location first...
#> Unzipping gt3x data to /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpG7zShD
#> 1/1
#> Unzipping /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T/RtmpG7zShD/PU3_CLE2B21130054_2017-03-16.gt3x
#>  === info.txt, log.bin extracted to /var/folders/1s/wrtqcpxn685_zk570bnx9_rr0000gr/T//RtmpG7zShD/PU3_CLE2B21130054_2017-03-16
#> GT3X information
#>  $ Serial Number     :"CLE2B21130054"
#>  $ Device Type       :"wGT3XPlus"
#>  $ Firmware          :"2.5.0"
#>  $ Battery Voltage   :"4.1"
#>  $ Sample Rate       :30
#>  $ Start Date        : POSIXct, format: "2017-03-16 16:30:00"
#>  $ Stop Date         : POSIXct, format: "2017-03-23 16:30:00"
#>  $ Last Sample Time  : POSIXct, format: "2017-03-23 16:30:00"
#>  $ TimeZone          :"01:00:00"
#>  $ Download Date     : POSIXct, format: "2017-03-27 13:16:37"
[... 222 lines cut ...]
#>  R.oo          1.23.0      2019-11-03 [1] CRAN (R 4.0.0)                   
#>  R.utils       2.9.2       2019-12-08 [1] CRAN (R 4.0.0)                   
#>  R6            2.4.1       2019-11-12 [1] CRAN (R 4.0.0)                   
#>  Rcpp          1.0.5       2020-07-06 [1] CRAN (R 4.0.0)                   
#>  read.gt3x   * 0.1.0.9000  2020-09-04 [1] Github (THLfi/read.gt3x@c4b83cc) 
#>  readr       * 1.3.1       2018-12-21 [1] CRAN (R 4.0.0)                   
#>  remotes       2.2.0       2020-07-21 [1] CRAN (R 4.0.2)                   
#>  rlang         0.4.7.9000  2020-08-25 [1] Github (r-lib/rlang@de0c176)     
#>  rmarkdown     2.3         2020-06-18 [1] CRAN (R 4.0.0)                   
#>  rprojroot     1.3-2       2018-01-03 [1] CRAN (R 4.0.0)                   
#>  sessioninfo   1.1.1       2018-11-05 [1] CRAN (R 4.0.0)                   
#>  stringi       1.4.6       2020-02-17 [1] CRAN (R 4.0.0)                   
#>  stringr       1.4.0       2019-02-10 [1] CRAN (R 4.0.0)                   
#>  testthat      2.99.0.9000 2020-08-25 [1] Github (r-lib/testthat@6a24275)  
#>  tibble        3.0.3       2020-07-10 [1] CRAN (R 4.0.2)                   
#>  usethis       1.6.1.9001  2020-08-25 [1] Github (r-lib/usethis@860c1ea)   
#>  vctrs         0.3.2       2020-07-15 [1] CRAN (R 4.0.2)                   
#>  withr         2.2.0       2020-04-20 [1] CRAN (R 4.0.0)                   
#>  xfun          0.16        2020-07-24 [1] CRAN (R 4.0.2)                   
#>  yaml          2.2.1       2020-02-01 [1] CRAN (R 4.0.0)                   
#> 
#> [1] /Library/Frameworks/R.framework/Versions/4.0/Resources/library
```

</details>

TuomoNieminen, 2020-11-01:
So I gather this:

- ACTIVITY1 packets are YXZ ordered
- ACTIVITY2 packets are XYZ ordered

However, ParseActivity() and ParseActivity2() treat the order identically. The parseGT3X() main function assumes XYZ order. Hence the activity1 type type is read incorrectly.

A solution is to either rename the columns in parseGT3X() based on activity type, or change ParseActivity() so that it switches the order of the first two colums.

muschellij2, 2020-11-01:
So the combination of https://github.com/THLfi/read.gt3x/pull/17/files#diff-d76857d9d7e56873c47c41e3f6e4efcfea6b0b033c0a5acaa80908755edc2108R500 and https://github.com/THLfi/read.gt3x/pull/17/files#diff-d76857d9d7e56873c47c41e3f6e4efcfea6b0b033c0a5acaa80908755edc2108R567 take care of this to my knowledge - the old Bin files and the ACTIVITY1

muschellij2, 2020-11-01:
Fixed and checked via https://github.com/THLfi/read.gt3x/pull/24

closed 2020-11-01 by muschellij2

muschellij2, 2020-11-09:
I think the issue is https://github.com/actigraph/GT3X-File-Format/blob/master/LogRecords/Activity2.md#activity-log-record-type-with-1-byte-payload can exist, but the test  flag indicates there's both ACTIVITY *and* ACTIVITY2 packets.

muschellij2, 2020-11-09:
Actually, more likely this: https://github.com/actigraph/GT3X-File-Format/blob/master/LogRecords/Activity.md#activity-log-record-type-with-1-byte-payload.  I found that it's likely the USB events that are messing this up.