THLfi/read.gt3x #22: Acceleration scale based on series name

angerhang, 2020-09-29:
Sometimes gt3x v2 might not have explicit acceleration scale in the `info.txt`, https://github.com/actigraph/GT3X-File-Format/blob/7f64246ef4033004ff20e7c399b4f00358550770/LogRecords/Activity2.md

So, one should set that according to the series sequence otherwise, we will have an error because the current implementation will stop when `acceleration scale` is missing from `info.txt`.

https://github.com/THLfi/read.gt3x/blob/c4b83cc262c7b3dc66b0408f9c72123686dd3e33/R/gt3x_info.R#L42

kels-2020, 2020-10-14:
Hi, I think I am experiencing the problem you describe. My info.text file does not contain acceleration scale information, like the sample gt3x file download with this package (EE_left_29.5.2017-05-30). 

I am getting an error:
 " Error in parseGT3X(logpath, max_samples = samples, scale_factor = info$`Acceleration Scale`,  : 
  Expecting a single value: [extent=0]. "

My info.text file
Serial Number: CLE1B49120284
Device Type: GT3XPlus
Firmware: 2.5.0
Battery Voltage: 4.2
Sample Rate: 30
Start Date: 636021288000000000
Stop Date: 0
TimeZone: 08:00:00
Download Date: 636033258360000000
Board Revision: 1
Unexpected Resets: 0

Is there away around this? 

Thanks,
Kelly

angerhang, 2020-10-14:
The java version I used sets the acceleration scale like this:
https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/0b3879386b06fc4bd41714b91de1ba59c34b892d/java/ActigraphReader.java#L581-L592

the official doc also suggests setting default acceleration scale [here](https://github.com/actigraph/GT3X-File-Format/blob/7f64246ef4033004ff20e7c399b4f00358550770/LogRecords/Activity2.md)

kels-2020, 2020-10-18:
Thanks for this.

muschellij2, 2020-11-01:
Can you upload the file - for any newer GT3X files, that shouldn't be missing unless it's corrupted.

muschellij2, 2020-11-20:
No movement on this issue and #26 has been presented as a potential fix.  Please reopen if it does not work.

closed 2020-11-20 by muschellij2