THLfi/read.gt3x #43: Timezone handling (enhancement suggestion)

vincentvanhees, 2022-08-15:
This is not an issue, but more of an enhancement suggestion.

For some studies we need to interpret timestamps in a different timezone then where the device was configured or the timezone of the system on which the package is run.

I am aware that:

> #' Read activity samples from a GT3X file as a matrix.
> #' Please note that all timestamps are in local time (of the device)
> #' even though they are represented as \code{POSIXct} with

So, I explored how I could facilitate the correct conversion of the timestamps. The only route I could find was:

```
# Input variables:
# configtz: time zone (database name) where the accelerometer was configured.
# desiredtz: time zone (database name)  where the accelerometer was worn.

timeChar = as.character(oldtime) #Strip the timestamp from it's timezone attribute by interpreting it as a character
time = as.POSIXct(timeChar, tz = configtz) # assigning time zone where the  accelerometer was configured.
attr(time, "tzone") <- desiredtz # Interpret the timestamp in the timezone of interest.
time = as.POSIXlt(time, tz = desiredtz) #Optionally change to POSIXlt format.
```

The challenge of doing these steps directly on the high resolution data inside read.gt3x is that the step `as.character` is very time consuming. It significantly slows down read.gt3x. Therefore, I decided to wait with timestamp conversion until I am working with epoch level aggregated data outside read.gt3x. There, I convert only the first timestamp and then extrapolate it to the rest of the time series based on known epoch length.

If anyone knows a way to convert the timestamps to the correct timezone without having to do the `as.character` operation in a fast way then maybe we could still add this as a new functionality to read.gt3x.

Alternatively, it may at least be helpful for others to include the above steps as a suggestion in the read.gt3x documentation? Happy to work on that if you agree.

TuomoNieminen, 2022-08-15:
The current datetime handling in read.gt3x is pretty confusing an improvements are welcome.

I would assume that lubridate::force\_tz is an efficient way of creating a new timestamp in a desired timezone, which I think is what we want.

A dummy example below. Say you have a timestamp created by read.gt3x which is stored as POSIXct UTC time but for which we know that the character representation is finnish local time, and so we want to force the correct timezone on it for further analysis.

``` r
# UTC timestamp which we here pretent to know to represent Finnish time in it's character representation.
x <- Sys.time()
attr(x, "tzone") <- "UTC"
print(x)
#> [1] "2022-08-15 12:46:59 UTC"
```

``` r
# force it to the correct timezone.
desiredtz <- "Europe/Helsinki"
newx <- lubridate::force_tz(x, tzone = desiredtz)
print(newx)
#> [1] "2022-08-15 12:46:59 EEST"
```

<sup>Created on 2022-08-15 by the [reprex package](https://reprex.tidyverse.org) (v2.0.1)</sup>

vincentvanhees, 2022-08-15:
Great, I can confirm that your solution is a lot faster than mine and speed is just a few seconds to do this for a week worth of data.

Would you like me to draft a PR for this? I may have some time this week.

TuomoNieminen, 2022-08-17:
Sure, please go ahead